YosysHQ/picorv32 #1: Unable to build testbench waveform in ubuntu

prashantrar, 2015-12-03:
Tried building the testbench.vcd in ubuntu 12.04 with vivado,

getting error as below.

prashantravi@ubuntu:~/picorv32/picorv32$ make testbench.vcd 
iverilog -o testbench.exe testbench.v picorv32.v
picorv32.v:41: syntax error
I give up.
make: *** [testbench.exe] Error 2
prashantravi@ubuntu:~/picorv32/picorv32$ 

Please help

dvc94ch, 2015-12-03:
I think you have modified the picorv32 makefile, it should look like this:

```
testbench.vcd: testbench.exe firmware/firmware.hex
    vvp -N $< +vcd
```

I assume you changed it because you don't have the riscv toolchain on your path. if you install the toolchain like is described in the README file, and make testbench.vcd should work just fine.
Undo your changes to your git branch with `git reset --hard && git checkout origin/master`.

cliffordwolf, 2015-12-03:
@prashantrar: It seems you have an older version of Icarus Verilog. Quoting the PicoRV32 README.md file:

> Note: The test bench is using Icarus Verilog. However, Icarus Verilog 0.9.7 (the latest release at the time of writing) has a few bugs that prevent the test bench from running. Upgrade to the latest github master of Icarus Verilog to run the test bench.

I think the recently released Icarus Verilog 10.0 should also work..

@dvc94ch: It's already the rule for testbench.exe that fails:

    testbench.exe: testbench.v picorv32.v
            iverilog -o testbench.exe testbench.v picorv32.v
            chmod -x testbench.exe

closed 2015-12-03 by cliffordwolf

prashantrar, 2015-12-07:
@cliffordwolf Thanks, the answer helped. it is working for me now. Just needed to download latest version of icarus verilog.