YosysHQ/picorv32 #7: scall instruction causes trap instead of interrupt

dvc94ch, 2016-05-14:
I'm working on porting a rtos to picorv. scall doesn't seem to be implemented even dough it's part of the user level isa. Is there a reason for this? I think this would fix the problem.

```
instr_sbreak <= !CATCH_ILLINSN && ((mem_rdata_q[6:0] == 7'b1110011 && (mem_rdata_q[31:7] == 'b0000000000010000000000000) || mem_rdata_q[31:7] == 'b0000000000000000000000000)) ||
					(COMPRESSED_ISA && mem_rdata_q[15:0] == 16'h9002));
```

cliffordwolf, 2016-05-15:
If the `CATCH_ILLINSN` parameter is set (the default), then SBREAK, as well as SCALL is simply decoded as illegal instruction and triggers IRQ 1 (assuming IRQs are also enabled of course, by default they are not!).

If IRQs are not enabled, then SBREAK and SCALL as well as any illegal instruction will cause a trap.

If `CATCH_ILLINSN` is not set, then SBREAK will trap the CPU, and SCALL and illegal instructions will have undefined behavior. The change you suggest would cause SCALL to also trap the CPU in this situation, but has no effect otherwise.

> I think this would fix the problem.

Think? Any reason why you did not try it?  ;)

I'm not sure why you are having problems with SCALL. I have just tried it using the [icosoc timer example](https://github.com/cliffordwolf/icotools/tree/master/icosoc/examples/timer). Adding `asm volatile ("scall" : : : "memory");` after `icosoc_maskirq(0);` will produce the following output:

```
Illegal instruction at 0x1b008: 0x00000073
pc  0x0001b00c    x8  0x00000000    x16 0x00000000    x24 0x00000000
x1  0x00010044    x9  0x00000000    x17 0x00000000    x25 0x00000000
x2  0x00030000    x10 0x0000e197    x18 0x00000000    x26 0x00000000
x3  0x0001d850    x11 0x00030004    x19 0x00000000    x27 0x00000000
x4  0x00000000    x12 0x00000000    x20 0x00000000    x28 0x00000000
x5  0x0001d130    x13 0x00000000    x21 0x00000000    x29 0x00000000
x6  0x0001d130    x14 0x00000000    x22 0x00000000    x30 0x00000000
x7  0x00000000    x15 0xffffffff    x23 0x00000000    x31 0x00000000
STOP.
```

Seems to be working just fine. Maybe you can provide more information on what you are trying to do, what behavior you would expect, and how the observed behavior differs from that. Until then I am now closing this issue.

closed 2016-05-15 by cliffordwolf

dvc94ch, 2016-05-16:
> Think? Any reason why you did not try it? ;)
Because in my infinite wisdom I decided to post something on the internet before hurrying to catch a train. =P

> An illegal instruction or bus error while the illegal instruction or bus error interrupt is disabled will cause the processor to halt.
Works exactly as documented...