YosysHQ/picorv32 #8: Incorrect handling of SBREAK/ILLNSN with PCPI

stevekerrison, 2016-05-31:
I'm trying to put a minimal example of this together (or a patch if I get there quicker), but won't get a chance until tomorrow, I leave this here for now as a test of my own sanity.

If PCPI is enabled and an illegal instruction or `sbreak` occur, upon `retirq`, the signal `pcpi_valid` remains asserted. The eventual effect of this, is that when the next valid PCPI instruction comes along, it causes another `ILLNSN` IRQ to occur, presumably because the PCPI timeout never got reset.

I actually want to handle `ecall`, and I guess I could do it with another PCPI extension, but that seems like a less graceful solution.

stevekerrison, 2016-06-01:
I have now tested this as a limited example in simulation, and in my normal use case on FPGA, and the behaviour is now as expected.

cliffordwolf, 2016-06-01:
In current git head ecall and ebreak now do not wait for the PCPI timeout. Also: q0 in IRQ handler (IRQ return address) now encodes if the interrupted instruction is a 16 or 32 bit opcode, simplifying decoding of the instruction that caused the IRQ. See firmware/irq.c handling of IRQ 2 (the block at "`if ((irqs & 6) != 0)`") for example code.

closed 2016-06-01 by cliffordwolf

stevekerrison, 2016-06-01:
That's great, thanks very much for the additional patch.

PR #9 (2016-06-01): Deassert pcpi_valid upon asserting sbreak IRQ
This fixes #8

It leaves two potentially unanswered problems/new issues:

- Instructions such as `sbreak` take at least 16 cycles to trigger an IRQ, as they wait for PCPI timeout. Writing logic to IRQ immediately on `sbreak` and potentially `scall` instructions might be desirable. Alternatively, an external PCPI module could do this, but would then have to use a different IRQ line.
- I haven't examined how an sbreak or illegal instruction would be handled inside the ISR, but trapping seems logical.
commit 38d51a33 (2016-06-01): Deassert pcpi_valid upon asserting sbreak IRQ

This fixes #8