YosysHQ/picorv32 #12: Possible waitirq stall

stevekerrison, 2016-06-23:
I've been debugging some IRQ related stuff I'm doing, and although this issue doesn't related to that, staring at this stuff did make me wonder.

The [`waitirq`](https://github.com/cliffordwolf/picorv32/blob/288a043acaa1db7cfda52c672afed32aa3ff4b0e/picorv32.v#L1234-L1239) instruction will sit and wait for an unmasked interrupt to take place, and then store the pending interrupt list into `rd`, so you can see what interrupts were serviced upon progressing.

However, `waitirq` is dependent on [`irq_pending`](https://github.com/cliffordwolf/picorv32/blob/288a043acaa1db7cfda52c672afed32aa3ff4b0e/picorv32.v#L1604), which is has the IRQ mask applied to it (obviously).

But what happens if:

1. IRQ is raised on a masked input.
2. I execute `maskirq` to unmask it.
3. My ISR is run.
4. I then execute `waitirq`.

Perhaps the interrupt I was expecting got serviced between `maskirq` and `waitirq`, which could happen if it raised any time before `waitirq` gets executed, right?

Is there a defect in my reasoning here, or would a stall be possible? If this sounds sane, I can try to write a simulation example that causes it.

cliffordwolf, 2016-06-23:
There seems to be a misunderstanding here what `waitirq` does. It waits until an IRQ is pending. The current mask set by `maskirq` has no relevance.

Note that `irq_mask` and `MASKED_IRQ` are two very different things: The former is a register holding the IRQ masked set by the last `maskirq` insn, and the latter is a *parameter* used to permanently masked out unused IRQ bits. An IRQ masked out by `MASKED_IRQ` does not trigger an IRQ and does not wake up the processor from the `waitirq` state. Only `MASKED_IRQ` is used for determining the value of `irq_pending`, not `irq_mask`.

The `make test` program even tests this for correct behavior. The processor start with all IRQs masked. The first thing the test prog does is waiting for a pending IRQ and only after that the IRQs are enabled:

https://github.com/cliffordwolf/picorv32/blob/288a043acaa1db7cfda52c672afed32aa3ff4b0e/firmware/start.S#L37

closed 2016-06-23 by cliffordwolf

stevekerrison, 2016-06-23:
Ah yes, what you described is the behaviour I *expected*, but as you rightly identified, I conflated `irq_mask` and the `MASKED_IRQ` parameter. Thanks!