YosysHQ/picorv32 #13: Opcode after jump executed on error

Edgar-Conzen, 2016-08-27:
To do some basic tests with your PicoRV32 I wrote a testbench, connected a rom with a simple
bare metal program in it and and let it run.

I started with a tiny C-program that I compiled with the riscv-gnu-toolchain

    #include <stdio.h>
    void main(void)
    { 
      register int cpureg15 asm ("a5") = 0; 
      //a5 = ABI-name of R15 
      
      while (1)
      {
        cpureg15++;
      }
    }

It compiled to :

      -- Test 1
    00000000 <main>:
       0:   ff010113            addi    sp,sp,-16
       4:   00812623            sw  s0,12(sp)
       8:   01010413            addi    s0,sp,16
       c:   00000793            li  a5,0
      10:   00178793            addi    a5,a5,1
      14:   ffdff06f            j   10 <main+0x10>  

[image]

The first thing I noticed was a xxxxxxxx on mem_addr[31:0] at about 150ns.
I NOP-ed the first three words since they were not required and the xxxxxxxx was gone. 

      -- Test 2
    00000000 <main>:
       0:   00000013            nop
       4:   00000013            nop
       8:   00000013            nop
       c:   00000793            li  a5,0
      10:   00178793            addi    a5,a5,1
      14:   ffdff06f            j   10 <main+0x10>  

[image]

I decided to go after this effect later on.

The next thing I saw were the TRAP at 275ns, mem_addr=0x00000018 at 205ns and 
mem_rdata =0x00000000 at 215ns.

As the riscv-spec points out on page 6 (Instruction length encoding) an opcode
of 0x00000000 shall lead to a trap. Since I indeed zeroed the unused portion
of the rom, I added a NOP at (the technically speaking unused) address 0x18 :
Now the TRAP was gone but register 15 stuck at zero and was obviously not incremented.
No wonder, since the two instructions that executed were at 0x14 and 0x18 (JUMP and NOP).

      -- Test 3
    00000000 <main>:
       0:   00000013            nop
       4:   00000013            nop
       8:   00000013            nop
       c:   00000793            li  a5,0
      10:   00178793            addi    a5,a5,1
      14:   ffdff06f            j   10 <main+0x10>  
      18:   00000013            nop

[image]

  
To ensure that the NOP at 0x18 was not only prefetched but also executed, I swapped the opcodes
of 0x10 and 0x18. Now R15 began incrementing. 

      -- Test 4
    00000000 <main>:
       0:   00000013            nop
       4:   00000013            nop
       8:   00000013            nop
       c:   00000793            li  a5,0
      10:   00000013            nop
      14:   ffdff06f            j   10 <main+0x10>  
      18:   00178793            addi    a5,a5,1

[image]

I swapped the opcodes back and calculated a jump further back, aiming at 0x0C.
In fact 0x10, 0x14 and 0x18 were executed and R15 was incrementing.

      -- Test 5
    00000000 <main>:
       0:   00000013            nop
       4:   00000013            nop
       8:   00000013            nop
       c:   00000793            li  a5,0
      10:   00178793            addi    a5,a5,1
      14:   ff9ff06f            j   0c <main+0x0c>  
      18:   00000013            nop

[image]

I replaced the NOP at 0x18 with a second increment and in fact R15 got incremented twice per loop.

      -- Test 6
    00000000 <main>:
       0:   00000013            nop
       4:   00000013            nop
       8:   00000013            nop
       c:   00000793            li  a5,0
      10:   00178793            addi    a5,a5,1
      14:   ff9ff06f            j   0c <main+0x0c>  
      18:   00178793            addi    a5,a5,1

[image]

My guess is, that the offset of the jump instruction is added correctly to the program counter
but only **after executing the opcode following the jump instruction by mistake**.

My question is : can you reproduce this effect or is something wrong in my setup?
Sadly I'm no verilog guy, so there is no use in trying to debug your code on my own.

Vivado 2016.1, ARTY-board (Artix-7 XC7A35T), riscv32-unknown-elf-gcc (GCC) 5.3.0

Project:
[picorv32test.zip](https://github.com/cliffordwolf/picorv32/files/440759/picorv32test.zip)

cliffordwolf, 2016-08-27:
Your main problem is that your memory asserts `mem_ready` all the time, signaling to the picorv32 core that `mem_rdata` already contains the read data, when in fact you block ram adds one cycle delay. Either fix your `mem_ready` handling or simply do an async read in `blockram7s.vhd`:

```
--  memrw : process (clk)
--  begin
--    if rising_edge (clk) then
--      dout <= memory (to_integer(unsigned(addr))); 
--    end if;
--  end process;

  dout <= memory (to_integer(unsigned(addr)));
```

Regarding the following instructions:

```
   0:   ff010113            addi    sp,sp,-16
   4:   00812623            sw  s0,12(sp)
   8:   01010413            addi    s0,sp,16
```

This is setting up a stack frame for `main()`. But because you are not using a libc (or any other startup code), your stack pointer is not set up (i.e. all 'x'). See the test firmware in `firmware/` in the picorv32 source package if you need some example code. If you only need to set up a stack pointer, you can also consider using the [STACKADDR](https://github.com/cliffordwolf/picorv32#stackaddr-default--32h-ffff_ffff) module parameter to picorv32 to set the stack pointer on reset.

closed 2016-08-27 by cliffordwolf

dvc94ch, 2016-08-27:
@wolf kicks ass =P thanks for maintaining all your awesome projects...

Edgar-Conzen, 2016-08-27:
Wow, that was fast! Thank you very much.