YosysHQ/picorv32 #25: synthesis mismatch with the latest rtl on xilinx spartan6 fpga

songchengwei, 2016-12-19:
Hi Clifford

Tried targeting the picorv32 to spartan6 (xc6slx25) board. The rtl simulation is ok, but FPGA didn't work, then I do a experiment to run the firmware you provided, it seems that the post-synthesis netlist have something wrong with the PC register, the PC add 0xC rather than 0x4 at certain point where I think there is  no jumb instruction.

Could you help to have a look at this, Thank you very much!

In waveform bad_wave_netlist_sim, the 62 (start from 1) mem_axi_araddr is 0x780, while the previous one is 0x774, in the good wavefrom the mem_axi_araddr is 0x778, this is the mismatch point.

The picorv32_netlist.txt is the post synthesis netlist.

The testbench, firmware.hex and rtl is the latest one, with id: ef86b30b2598df6993a473222ebe355b0348e226

[image]
[image]

[picorv32_netlist.txt](https://github.com/cliffordwolf/picorv32/files/660877/picorv32_netlist.txt)

cliffordwolf, 2016-12-19:
Possible duplicate of #2.

Generally speaking: XST has a couple of bugs and Xilinx does not fix them because you should use Vivado for new designs anyways (I know that Vivado does not support 6-series devices, I'm just paraphrasing the position of Xilinx).

I don't see it as my responsibility to work around strange bugs in obsolete software..

If you have to use Xilinx 6-series devices, and don't have access to e.g. Synplify Pro for Xilinx synthesis, and therefore are locked to XST as implementation tool, I'd recommend using Yosys to convert the PicoRV32 Verilog code into simpler HDL code before passing it to XST for implementation.

Something like the following Yosys command line should do the job:

    yosys -p 'prep -top picorv32; write_verilog -noattr picorv32_presyn.v' picorv32.v

closed 2016-12-19 by cliffordwolf

songchengwei, 2016-12-20:
Hi, Clifford

I use yosys to convert the behavior picorv32 model to rtl level description, still the same issue.

I'll try to switch to other device and use vivado to synthesis the core, thank you for your help!