YosysHQ/picorv32 #29: Issue while Building the RV32I Toolchain

shuvamnandi, 2017-01-16:
Hi, 
I was trying to build the RV32I toolchain and got an error at this step "../configure --with-arch=RV32I --prefix=/opt/riscv32i":

`../configure --with-arch=RV32I --prefix=/opt/riscv32ichecking for gcc... gccchecking whether the C compiler works... yeschecking for C compiler default output file name... a.outchecking for suffix of executables...checking whether we are cross compiling... nochecking for suffix of object files... ochecking whether we are using the GNU C compiler... yeschecking whether gcc accepts -g... yeschecking for gcc option to accept ISO C89... none neededchecking for grep that handles long lines and -e... /bin/grepchecking for fgrep... /bin/grep -Fchecking for grep that handles long lines and -e... (cached) /bin/grepchecking for bash... /bin/bashchecking for __gmpz_init in -lgmp... yeschecking for mpfr_init in -lmpfr... yeschecking for mpc_init2 in -lmpc... yeschecking for curl... /usr/share/centrifydc/bin/curlchecking for wget... /usr/bin/wgetchecking for ftp... /usr/bin/ftpconfigure: error: Unknown arch`

Did anyone else also face this problem?

cliffordwolf, 2017-01-16:
Oops. I've updated to a new toolchain version yesterday and forgot to update that bit of documentation. The arguments for `--with-arch=` is now lowercase. Fixed README in e9b6bcf. Thanks for reporting this issue.

closed 2017-01-16 by cliffordwolf

shuvamnandi, 2017-01-19:
Yes, it works after changing the argument. Thanks!