YosysHQ/picorv32 #33: jal instruction execution doesn't align with expectation

songchengwei, 2017-02-21:
Hi, Clifford

When I simulate picorv32 with my system, the jal instruction does't jump the the expected address.
[image]

as you can see from the waveform, when the current pc is 0x512, cpu get a jal instruction, the value is 0xbfdd in the objdump:

512: bfdd     j   508 <main+0x8>, 

since main starts at 0x500, I expect the pc jump to 0x508, however, it jumps to 0x1508, the variable decoded_imm_uj is 0xff6, when it adds to 0x512, the result is 0x1508, should we just use the lower 12 bits to generate reg_next_pc?  

Could you please have a look at this? Thank you very much!

log.txt is the simulation log when add DEBUY macro for picorv32
mem.txt is the memory initialization file

[log.txt](https://github.com/cliffordwolf/picorv32/files/789630/log.txt)
[mem.txt](https://github.com/cliffordwolf/picorv32/files/789629/mem.txt)

cliffordwolf, 2017-02-21:
I cannot reproduce this, as you can see on this screenshot:

[image]

This is made using the default PicoRV32 `make test` with the following modification to `start.S` so it executes a `0xbfdd` instruction at address `0x512`. In my test execution continues at `0x508` as expected.

```
diff --git a/firmware/start.S b/firmware/start.S
index 06d4744..3264752 100644
--- a/firmware/start.S
+++ b/firmware/start.S
@@ -337,7 +337,15 @@ irq_stack:
 /* Main program
  **********************************/
 
+.balign 0x400
+.space 0x108
+start_2:
+j start_1
+.space 8
 start:
+.hword 0xbfdd
+
+start_1:
 	/* zero-initialize all registers */
 
 	addi x1, zero, 0
```

Can you provide a full test bench that reproduces the issue you are seeing?

songchengwei, 2017-02-22:
Hi, Clifford 

Thank you for your time!

I have the waveform attached, the picorv32 rtl is the latest version, with git id c7cc32.

Is it enough for you to debug? If not, I think I can provide a simple env to recreate the scenario and provide all the source code. 

[testbench.tar.gz](https://github.com/cliffordwolf/picorv32/files/792024/testbench.tar.gz)

songchengwei, 2017-02-23:
Hi, Clifford

I use a simple testbench to reproduce the scenario, there is no io, cpu is the only dut. All the related files are packed in the attached file.

As you can see in the waveform, when jump instruction is received, the reg_pc jump from 0x4f0 to 0x14e6, where I think it should jumb to 0x4e6.

Could you please have a look at this?

[image]

[picorv32.tar.gz](https://github.com/cliffordwolf/picorv32/files/795866/picorv32.tar.gz)

cliffordwolf, 2017-02-25:
I think this is a bug in VCS. I ran your code in iverilog and it looks fine here:

[image]

Unfortunately I do not have VCS, so I cannot track down the bug. But I would guess this will trigger it:

```
module test;
  reg [31:0] decoded_imm_uj;
  reg [31:0] mem_rdata_latched = 'hbfdd;
  initial begin
    { decoded_imm_uj[31:11], decoded_imm_uj[4], decoded_imm_uj[9:8], decoded_imm_uj[10], decoded_imm_uj[6],
      decoded_imm_uj[7], decoded_imm_uj[3:1], decoded_imm_uj[5], decoded_imm_uj[0] } = $signed({mem_rdata_latched[12:2], 1'b0});
    $display("%x", decoded_imm_uj);
    $finish;
  end
endmodule
```

The correct output is `fffffff6`. I suspect VCS will output `00000ff6`.

cliffordwolf, 2017-02-26:
I asked around on twitter if anyone with access to VCS can reproduce this. I got a reply from someone who would like to stay anonymous who has access to VCS 2014.03.  He ran your test bench in his VCS, but his results do not match yours. They match however what I get when I run your code in icarus verilog:

[image]

He had to make minor modifications to your makefile in order to be able to run it on his system:

[image]

I'm assuming your VCS version is different from his. Depending on which is newer, I'd say they either fixed or introduced that bug between the two versions..

cliffordwolf, 2017-03-14:
OP inactive for >2 weeks. Closing issue.

closed 2017-03-14 by cliffordwolf

c60k28, 2018-02-10:
OP test, reg PC goes from h4f0 to h4e6:
[image]

Version:
Version M-2017.03-SP2-1_Full64

cliffordwolf, 2018-02-10:
Thanks @c60k28. I'd say that means the VCS bug is fixed in newer versions of the tool.

renli093, 2019-02-13:
My vcs version is vM-2017.03-SP2 and meet the similar bug in simulation. 
If add signed in decoded_imm_uj define, the simulation can pass.

	reg signed [31:0] decoded_imm_uj;