YosysHQ/picorv32 #38: Xilinx ISE 14.7 synthesis

pmonta, 2017-03-15:
A few changes are needed for successful synthesis by the legacy Xilinx ISE toolchain:

- xst incorrectly implements the register file with ENABLE_REGS_DUALPORT = 1, but ENABLE_REGS_DUALPORT = 0 works fine
- xst doesn't like memory in an always @* combinational block
- xst has some problems with parameterized macros

I did look at issues #2 and #25, and I'm a little puzzled as to how synthesis even completed, unless they made the same changes to the RTL as I did.  I haven't seen problems with the PC register once the design is in runnable shape.

Attached is an example project targeting the Spartan 3E Starter board.  It contains a picorv32 core with the plain memory interface, some block RAM, a UART, and some test software in C.

I'm just putting this out there in the hope it will make picorv32 more useful for older Spartan-6 or Spartan-3E designs; the main repository probably doesn't want to deal with irksome `ifdef OLD_XILINX stuff.  I have some Spartan-6 hardware for which I'd like to use picorv32 (currently using picoblaze).

[picorv32-Xilinx-ISE.tar.gz](https://github.com/cliffordwolf/picorv32/files/845646/picorv32-Xilinx-ISE.tar.gz)

luismarques, 2017-03-26:
@pmonta This explains a lot. I thought I had a bug in my cache implementation. Eventually I reduced the problem to the picorv32, and that's when I did a Google search that found this issue. Thanks for providing your fixes and the other files!

(I'm using a Papilio Pro, with a Xilinx Spartan 6 LX9. Too bad the Spartan 6 isn't supported by the newer Xilinx tools, and they have stopped maintaining the old ones...)

cjhonlyone, 2020-04-11:
you shoud do this to prevent cpuregs to be synthesised to block RAM
	(* ram_style = "distributed" *)
	reg [31:0] cpuregs [0:regfile_size-1];

armleo, 2020-07-25:
For always @* try using xst new parser