YosysHQ/picorv32 #41: Make fails

riscveval, 2017-07-04:
Hi when i am trying to Run make test or any make commands its produce the following error   picorv32.v:1878: error: invalid module item.
picorv32.v:1879: syntax error
picorv32.v:1879: error: Invalid module instantiation
picorv32.v:1880: error: Invalid module instantiation
picorv32.v:1881: error: Invalid module instantiation
picorv32.v:1885: error: invalid module item.
picorv32.v:1887: syntax error
can you give guidance to correct the above mentioned issue.

cliffordwolf, 2017-07-04:
> can you give guidance to correct the above mentioned issue.

Quoting the README file:

> Note: The test bench is using Icarus Verilog. However, Icarus Verilog 0.9.7 (the latest release at the time of writing) has a few bugs that prevent the test bench from running. Upgrade to the latest github master of Icarus Verilog to run the test bench.

closed 2017-07-04 by cliffordwolf

ghanesimit, 2019-12-31:
Is there any complete guide which I can follow to compile or test this source on board?

fatalfeel, 2021-09-27:
here is example 
http://fatalfeel.blogspot.com/2021/09/risc-v-on-icesugar.html