YosysHQ/picorv32 #45: cpuregs[0] is not hard-wired zero

markusdd, 2017-07-13:
I recently observed in simulation, that in contrast what the ISA spec demands, register 0 is not a hard-wired 0. It is writeable just like any other register, which of course can cause problems with certain instructions and dummy instructions that expect a zero to be there (essentially it is meant to be a /dev/null ). It can be observed when using an unconditional jump, e.g. of the form JAL x0,8. The return address will then be in x0, although it should remain 0. 

A possible fix could be changing

always @(posedge clk) begin
    if (resetn && cpuregs_write)
      cpuregs[latched_rd] <= cpuregs_wrdata;
  end

into

always @(posedge clk) begin
    cpuregs[0] <= 32'h0000_0000;
    if (resetn && cpuregs_write && |latched_rd)
      cpuregs[latched_rd] <= cpuregs_wrdata;
  end

cliffordwolf, 2017-07-13:
Obviously this "bug report" is complete nonsense. Of course x0 = 0 is enforced: It's implemented on the register file read port just a few lines below the code you have quoted. Your proposed "fix" would not change the behavior of the core at all, but it would of course break instantiation of dedicated memory resources, for example in FPGA flows.

If you make statements like "It can be observed when using an unconditional jump, e.g. of the form JAL x0,8." then you should better produce an example program that actually demonstrates that the core violates the ISA spec, not just make a false statement about it. But of course you can't, because the core doesn't.

Do you really think the core would pass the insn tests from riscv-tests, pass riscv-torture, pass csmith+gcc based random testing, and pass formal verification, if x0 wouldn't be a zero register?! Think again!

closed 2017-07-13 by cliffordwolf

markusdd, 2017-07-13:
Why are you getting so personal?

I have a little stupid example program and in the waveforms it can be seen when looking at cpuregs[0]:
0x00000013
0x00000013
0x01E00E93
0x02800F13
0xFFDFF06F

[image]

The reason the tests all pass is simply because you are masking it at the read port, but the Flops/Memory get manipulated nonetheless, which is not ideal (e.g. from a power perspective). This is 2 sides of the same medal. Because not everyone (me included) intends to use memory for that regfile (especially when targeting an ASIC, memory is way to expensive and slow) but rather flops, and when you e.g. connect debugging resources to the regfile registers or implement custom instructions that rely on them, you might observe ghost values that never actually can be read via the regular ports.
Alternatively one could scrap the implementation of cpuregs[0] altogether. This should not break the memory inference if it is desired.

So please, I was trying to contribute and be constructive. It is ok when you disagree with my particular solution or views, but being respectful is not too much to ask for (and calling it non-sense from the get-go without knowing what I did is disrespectful.). 

Anyway, I want to thank you for releasing this core here, it is definitely one of the best freely available RISC-V Implementations. 
Feel free to tell me if you want  me to contribute further stuff, e.g. I would have an AHB-Lite Master Interface for the core if you're interested. I can also just fork the project, but I felt contributing to the mainline is probably more valuable.

cliffordwolf, 2017-07-13:
> Why are you getting so personal?

Sorry. I take your bug report personal because you accuse me of not checking for the most trivial thing in my implementation, while at the same time you don't even bother producing a test case that demonstrates that there is an issue. If you had bothered to do so, you would have seen that there is no bug.

> looking at cpuregs[0]

cpuregs[0] is not observable through the ISA. It's not relevant what value is there. The only relevant thing is the externally observable state. For example, provide a code fragment that makes a compliant implementation write value 0 to memory address 12345678, but PicoRV32 writes 1 instead.

Would you also claim that an out-of-order riscv implementation violates the RISC-V spec by executing the instructions in the wrong order? No! Because it does not matter what the core does internally as long as the externally observable behavior is correct.

> The reason the tests all pass is simply because you are masking it at the read port

You frame it like it passes the tests but is not ISA compliant. That's nonsense! It is ISA compliant. The ISA spec only says what the externally observable behavior is, not how that must be achieved in the processor microarchitecture.

It is ISA compliant and therefore there is no bug because I am masking it at the read port.

> which is not ideal (e.g. from a power perspective)

sure. But (1) preventing a synthesis tool from using dedicated resources for the register is is much much worse, and not only for power, and (2) the constant assignment `cpuregs[0] <= 32'h0000_0000;` is completely unnecessary.

So there is a power argument for adding `|latched_rd` to that if condition. Ok. Evaluate it and let me know what the area/power tradeoff is.

But your claim was that the core is not ISA compliant. And that's obviously bogus, and making such a claim is just a waste of everyones time if you do not bother to create a test cases that shows that there is a problem. And again: The ISA spec only specifies externally visible behavior, so it does not matter what value some internal signal has.

And before you make an argument that this was all about power from the get-got: Please read your original post and point out where exactly you even mention an issue with power.

> non-sense from the get-go without knowing what I did is disrespectful

The claim is non-sense and I have now explained a 2nd time why that is the case.

You wrote "which of course can cause problems with certain instructions and dummy instructions that expect a zero to be there". Either you are very bad at expressing what you mean (in which case I'm truly sorry), or you wrote this assuming that an instruction reading x0 would read non-zero value (at least in some cases), which is obviously false and shows that you did not bother checking your claim before posting.

I mean no disrespect. Really. But I don't think calling your claim out as what it is is any more rude than claiming that I failed to check the most basic aspect of the RISC-V ISA and not even bothering providing a simple test case that demonstrates that claim.. (I'm not offended by the implication that there is a bug. I'm offended by the implication that my time worth so little that it is ok to waste it by reporting a bug without even considering writing a test case first.)

> Anyway, I want to thank you for releasing this core here, it is definitely one of the best freely available RISC-V Implementations.

Thanks.

markusdd, 2017-07-13:
> You wrote "which of course can cause problems with certain instructions and dummy instructions that expect a zero to be there". Either you are very bad at expressing what you mean (in which case I'm truly sorry)

I don't want this to be an argument thread, because I appreciate your work, maybe it becomes more obvious when I explain shortly what I tried to do and where I ran into problems:

I did an FPGA trial mapping. As there is no JTAG in-core debugging, I went the quick route and memory mapped the registers by putting an APB decoder on the outside, directly wiring the registers to it as read-only from the bus, because I have this hardware readily available. So I could observe the internal core state from the bus while it was running. To my surprise, Reg 0 changed values although software executed fine. Then I ran the simulation and understood why. At this point, I did not assume you wanted the regs always to be mapped to FPGA memory, so I really wondered why you would write it this way.

So let me make my "claim" more precise: The software execution is a 100% fine and compliant, It just never crossed my mind as an implementation option that the reg would change values and just be masked at the output. That let me on the road reporting this, without meaning any harm (hence the title cpuregs[0] is not hard wired to zero, which is true, but it is seen as 0 because of the read logic).

Also it is 100% correct that the external state is compliant, but it gets tricky when you try to implement internal state debugging and helper instructions (which is what I meant with certain instrucutions and dummy instructions, I was looking into how to extend the core in that way). I'll admit that was pretty misleading now looking back at it.

I hope we're good now, just leave it as it is. I will remove reg0 from the observable registers in my system, this should take care of it for now.

markusdd, 2017-07-13:
decided to remove everything, as this might confuse people and create mistrust in the implementation, which would not be appropriate.

Sorry.