YosysHQ/picorv32 #46: io constraints

kunalg123, 2017-07-22:
Hi Clifford,
I am trying to implement and STA this block using opensource qflow and opentimer. Can you please help me know what are the IO constraints that needs to be used? Any pointer or pdf will help....I am using target clock frequency of 400MHz to start with....
Thanks

cliffordwolf, 2017-07-22:
Hmm.. this looks more like a qflow+opentimer question than something related to PicoRV32 directly. Unfortunately I'm no expert in either. I'll poke Tim from qflow, maybe he has some pointer for you.

kunalg123, 2017-07-22:
I see
Actually, I was looking for values for input/output port delay values for picorv32. I will check with Tim. Thanks

closed 2017-07-22 by kunalg123

RTimothyEdwards, 2017-07-22:
I'll work with Kunal directly on this, thanks.