YosysHQ/picorv32 #48: problems about the docs

woodyplum, 2017-08-28:
hi, 
   when study the picorv32, some confused without documents, where can i find the descriptions or request for help on these?  
1. about the interrupt vector table description, from the De-assambled  code of the example, seems the ram 0-F both for the reset handler, how should i arrange the vectors, or just like the one in original riscv? 

2. is there any Unit for Debug implementation(i mean the support for breakpoint, step...)? it should important for the fpga debug, while can not find in the readme.

3. any PCPI detail descriptions? can not find the modules in the readme for pcpi functions, while the PCPI interface always toggle even the parameter PCPI closed. 

thanks.

cliffordwolf, 2017-08-28:
> about the interrupt vector table description, from the De-assambled code of the example, seems the ram 0-F both for the reset handler, how should i arrange the vectors, or just like the one in original riscv?

I cannot parse this. What do you mean?

> is there any Unit for Debug implementation(i mean the support for breakpoint, step...)? it should important for the fpga debug, while can not find in the readme.

You don't need any extra hardware for this (I assume this is what you mean with "unit").

The core execute the illegal instruction trap handler for EBREAK.

You have to write your own gdb stub, I don't provide one.

> any PCPI detail descriptions?

There is a chapter in the readme file about the PCPI.

> can not find the modules in the readme for pcpi functions, while the PCPI interface always toggle even the parameter PCPI closed.

I have no idea what that means.

woodyplum, 2017-08-29:
about the interrupt vector table description, from the De-assambled code of the example, seems the ram 0-F both for the reset handler, how should i arrange the vectors, or just like the one in original riscv?

I cannot parse this. What do you mean?

<<what i mean is where should I place the reset entry address and each ISR entry address in the memory?  for example in ARM system the 0x0 is for SP, 0x4 is for reset handler and so on.>>

    is there any Unit for Debug implementation(i mean the support for breakpoint, step...)? it should important for the fpga debug, while can not find in the readme.

You don't need any extra hardware for this (I assume this is what you mean with "unit").

The core execute the illegal instruction trap handler for EBREAK.

You have to write your own gdb stub, I don't provide one.
<<can not understand  this, RISCV debug specification should need dedicated HW to support, like PC address compare, step exec.., like coresight in ARM.>>

    any PCPI detail descriptions?

There is a chapter in the readme file about the PCPI.

    can not find the modules in the readme for pcpi functions, while the PCPI interface always toggle even the parameter PCPI closed.

I have no idea what that means.
 <<i try the basic testcase simulation, in the result waveform dumped, the interface for the PCPI input all floating while PCPI output keep toggle, and can not see the connections for the PCPI co-processors, any case for test this implementation? >>

thanks.

cliffordwolf, 2017-08-30:
> what i mean is where should I place the reset entry address and each ISR entry address in the memory? for example in ARM system the 0x0 is for SP, 0x4 is for reset handler and so on.

To whatever you have set the PROGADDR_RESET and PROGADDR_IRQ of course. Theyt are both documented in the README file.

> can not understand this, RISCV debug specification should need dedicated HW to support

PicoRV32 does not support the interfaces described in the debug specification.

> i try the basic testcase simulation, in the result waveform dumped, the interface for the PCPI input all floating while PCPI output keep toggle, and can not see the connections for the PCPI co-processors, any case for test this implementation?

(1) What do you care if some of the output pins are toggling if you have disabled the interface? Presumably you are not using them anyways (because you have disabled it), and as long as `pcpi_valid` is low anything connected to the port should ignore all the other lines anyways.

(2) Re tests: e.g. MUL/DIV are implemented using PCPI.

closed 2017-08-30 by cliffordwolf