YosysHQ/picorv32 #50: riscv-gcc commit does not exist?

drichmond, 2017-11-09:
Hi Clifford - 

Tried running your makefile / build scripts for the RISC-V GCC toolchain and got the following error on two computers:

```make
Cloning into 'riscv-gcc'...
Checking connectivity... done.
fatal: reference is not a tree: 4fb4d8f9e9ac8a28d6ea5117688eadbcd0f7978e
Unable to checkout '4fb4d8f9e9ac8a28d6ea5117688eadbcd0f7978e' in submodule path 'riscv-gcc'
make[1]: *** [build-riscv32im-tools-bh] Error 1
make[1]: Leaving directory `/home/drichmond/Research/repositories/git/picorv32'
make: *** [build-riscv32im-tools] Error 2
```

Perhaps the commit tree changed recently? I was able to compile this code using your instructions several days ago.

wallento, 2017-11-09:
Yes, I can confirm the bug. Seems the referenced submodule commit was removed, maybe a rebase or related to this: https://github.com/riscv/riscv-gnu-toolchain/commit/28db62c566d3bbb83ad791d7df88fd09f73dc269

Changing the following line in the Makefile should help:

```
diff --git a/Makefile b/Makefile
index 6622dcc..892b520 100644
--- a/Makefile
+++ b/Makefile
@@ -1,5 +1,5 @@
 
-RISCV_GNU_TOOLCHAIN_GIT_REVISION = e9f5458
+RISCV_GNU_TOOLCHAIN_GIT_REVISION = b4dae89
 RISCV_GNU_TOOLCHAIN_INSTALL_PREFIX = /opt/riscv32
 
 SHELL = bash
```

cliffordwolf, 2017-11-18:
Unfortunately I get linker errors with the new version of the tools. That's why I did not update yet. So far I did not find the time to have a closer look. If any of you two would like to have a look: please do that.

drichmond, 2017-11-18:
Currently compiling head - what errors do you see?

cliffordwolf, 2017-11-18:
@drichmond Simple run `make firmware/firmware.hex`. Will generate a "6 bytes required for alignment to 8 bytes boundary" message. Probably just an easy fix in one of the asm files I copied from riscv-tests. I did not have time to look into it so far.

wallento, 2017-11-18:
I can have a look at it later tonight and prepare the PR. Cheers

cliffordwolf, 2017-11-18:
Seems like `.balign` is broken.

This is `tests/riscv_test.h`:

```
        jal     zero,.prname_next;      \
.test_name:                             \
        .ascii TEST_FUNC_TXT;           \
        .byte 0x00;                     \
        .balign 4;                      \
.prname_done:                           \
        addi    a1,zero,'.';            \
```

And this is `riscv32-unknown-elf-objdump -d tests/auipc.o`:

```
  1c:	ff1ff06f          	j	c <.prname_next>

00000020 <.test_name>:
  20:	7561                	lui	a0,0xffff8
  22:	7069                	0x7069
  24:	          	beqz	zero,24 <.test_name+0x4>

00000026 <.prname_done>:
  26:	02e00593          	li	a1,46
```

The symbol `.prname_done` should be aligned to a multiple of 4 bytes. But for some reason it isn't.

cliffordwolf, 2017-11-18:
Filed an issue for riscv-gnu-toolchain: https://github.com/riscv/riscv-gnu-toolchain/issues/298

cliffordwolf, 2017-11-19:
Pushed a toolchain update and fix for the `.balign` issue in commit df0113218550838fd2ba1ec6cdb9f24499e83cff.

closed 2017-11-19 by cliffordwolf