YosysHQ/picorv32 #51: riscv32imac

fathomtreelimited, 2017-11-12:
Hi!

Linux RISCV / musl is making it mandatory to have atomic instructions.  

Can you see A support being added to picorV32?

Thanks.

cliffordwolf, 2017-11-13:
For the most part atomics are not a cpu feature but a feature of the memory subsystem. What memory interface would you suggest that would support atomics and would not require more resources then the entire processor core itself?

Would you be using uC-Linux for this? (PicoRV32 is of course MMU-less. Just an MMU stand-alone would probably be already larger than the whole PicoRV32 core.)

Also: What multi-core application would that be exactly where PicoRV32 would be a good choice? Since PicoRV32 is not a scalar architecture: Wouldn't it make more sense to switch from one PicoRV32 core to one scalar core, instead of using multiple PicoRV32 cores?

closed 2017-11-13 by cliffordwolf

fathomtreelimited, 2017-11-13:
I have considered adding a PCPI for the 11 "A"  instructions in the spec.

LR
SC
AMOADD
AMOXOR
AMOOR
AMOAND
AMOMIN
AMOMAX
AMOMINU
AMOMAXU
AMOSWAP

minexew, 2018-03-18:
@cliffordwolf Thanks for keeping this small and simple!

DmitryZlobec, 2025-04-29:
> For the most part atomics are not a cpu feature but a feature of the memory subsystem. What memory interface would you suggest that would support atomics and would not require more resources then the entire processor core itself?
Your right, but "A" subsystem really needed for supporting baremetall Rust. Without LR/SC impossible to make mutex. but mutex is really need for rust memory model. It is a feature of LLVM , and I could not find solution to make mutex software  without LR/SC.
May be it can be made as simple LW without multiprocessing for unithread application?