YosysHQ/picorv32 #55: Is there a way to simulate DRAM memory latencies with this repository?

robsmi1974nyc, 2018-02-16:
It would be interesting to do that to get more realistic performance metrics, even if a simplistic 100 cycle delay to get to main memory.

frantony, 2018-02-16:
Please see https://github.com/open-design/riscv-soc-cores/blob/master/cores/picorv32-wb-soc/bench/picorv32_wb_soc_tb.v.

picorv32_wb_soc uses picorv32 with Wishbone connector and wb_sdram_ctrl.

cliffordwolf, 2018-02-16:
PicoRV32 has no caches. If you have a 100 cycles memory latency then it will take 100 cycles for a single instruction (200 cycles for memory load, memory store, and branches taken).

If you want reasonable performance with DRAM then you will need to write your own cache IP and a testbench for that which you can use to simulate performance and check that it is working correctly.

closed 2018-02-16 by cliffordwolf