YosysHQ/picorv32 #71: Verilator generated executable didn't run far...

dragonware-com, 2018-05-24:
Using the following verilator commands:
`verilator -Wno-lint -Wno-MULTIDRIVEN -trace --top-module picorv32_wrapper --cc testbench.v picorv32.v --exe testbench.cc
cd obj_dir/
make -j -f Vpicorv32_wrapper.mk
`
Got these error message:
`TRAP after 8428 clock cycles
ERROR!
%Error: testbench.v:268: Verilog $stop
Aborting...
Aborted (core dumped)
`

cliffordwolf, 2018-05-25:
You likely built a test firmware using compressed ISA. But you did not pass `-DCOMPRESSED_ISA` to verilator, so you built a simulation model without compressed ISA support. Therefore the core hangs when it encounters the first compressed RISC-V instruction.

Solutions: Either build the simulation model with `-DCOMPRESSED_ISA` or build the firmware image with `COMPRESSED_ISA=` (instead of `COMPRESSED_ISA=C`).

I've now added a `make test_verilator` target in commits 247a19d and 3f55fb4 that builds and runs the verilator testbench.

closed 2018-05-25 by cliffordwolf

dragonware-com, 2018-05-25:
Yeap, that's it. Thanks Clifford.  Just a typo in the Makefile:

```
diff --git a/Makefile b/Makefile
index 2c14263..715249e 100644
--- a/Makefile
+++ b/Makefile
@@ -74,7 +74,7 @@ testbench_verilator: testbench.v picorv32.v
        verilator --cc --exe -Wno-lint -trace --top-module picorv32_wrapper testbench.v picorv32.v testbench.cc \
                        $(subst C,-DCOMPRESSED_ISA,$(COMPRESSED_ISA)) --Mdir testbench_verilator_dir
        $(MAKE) -C testbench_verilator_dir -f Vpicorv32_wrapper.mk
-       cp obj_dir/Vpicorv32_wrapper testbench_verilator
+       cp testbench_verilator_dir/Vpicorv32_wrapper testbench_verilator
```

dragonware-com, 2018-05-25:
Another bug in the showtrace.py is probably also related to "compressed_isa". 
```
0000ac44 <tdat>:

    ac44:	          	jal	t4,15c44 <end+0xaf90>

Traceback (most recent call last):
  File "showtrace.py", line 16, in <module>
    if match: insns[int(match.group(1), 16)] = (int(match.group(2), 16), match.group(3).replace("\t", " "))
ValueError: invalid literal for int() with base 16: 'jal'
```

cliffordwolf, 2018-05-25:
Thanks! 1cc9784 fixes the makefile typo, 4fed27f fixes showtrace (had nothing to do with compressed_isa).