YosysHQ/picorv32 #76: Write Address and Data undefined with simple hello world program

bentomo, 2018-07-30:
I'm attempting to write a simple hello world function to run on picorv32 in sim. I'm running in the scripts cxxdemo folder. I can run **make test** just fine, but when I try to write a custom Hello World example I get an undefined write address and data: WR: ADDR=xxxxxxxX DATA=xxxxxxxx MASK=1111

**hello.cc**
```
 #include <stdio.h>
 #include <iostream>
 #include <vector>
 #include <algorithm>

 int main() {

	printf("Hello World!\n");

	return 0;
 }
```

**Makefile** (modified of cxxdemo makefile)
```
RISCV_TOOLS_PREFIX = /opt/riscv32ic/bin/riscv32-unknown-elf-
CXX = $(RISCV_TOOLS_PREFIX)g++
CC = $(RISCV_TOOLS_PREFIX)gcc
AS = $(RISCV_TOOLS_PREFIX)gcc
CXXFLAGS = -MD -Os -Wall -std=c++11
CCFLAGS = -MD -Os -Wall -std=c++11
LDFLAGS = -Wl,--gc-sections
LDLIBS = -lstdc++

test: testbench.vvp firmware32.hex
	vvp -N testbench.vvp

hello: testbench_hello.vvp hello32.hex
	vvp -N testbench_hello.vvp

testbench.vvp: testbench.v ../../picorv32.v
	iverilog -o testbench.vvp testbench.v ../../picorv32.v
	chmod -x testbench.vvp

testbench_hello.vvp: testbench_hello.v ../../picorv32.v
	iverilog -o testbench_hello.vvp testbench_hello.v ../../picorv32.v
	chmod -x testbench_hello.vvp

firmware32.hex: firmware.elf start.elf hex8tohex32.py
	$(RISCV_TOOLS_PREFIX)objcopy -O verilog start.elf start.tmp
	$(RISCV_TOOLS_PREFIX)objcopy -O verilog firmware.elf firmware.tmp
	cat start.tmp firmware.tmp > firmware.hex
	python3 hex8tohex32.py firmware.hex > firmware32.hex
	rm -f start.tmp firmware.tmp

hello32.hex: hello.elf
	$(RISCV_TOOLS_PREFIX)objcopy -O verilog hello.elf hello.tmp
	cat hello.tmp > hello.hex
	python3 hex8tohex32.py hello.hex > hello32.hex
	rm -f hello.tmp

firmware.elf: firmware.o syscalls.o
	$(CC) $(LDFLAGS) -o $@ $^ -T ../../firmware/riscv.ld $(LDLIBS)
	chmod -x firmware.elf

hello.elf: hello.o
	$(CC) $(LDFLAGS) -o $@ $^ -T ../../firmware/riscv.ld $(LDLIBS)
	chmod -x hello.elf

start.elf: start.S start.ld
	$(CC) -nostdlib -o start.elf start.S -T start.ld $(LDLIBS)
	chmod -x start.elf

clean:
	rm -f *.o *.d *.tmp start.elf
	rm -f firmware.elf firmware.hex firmware32.hex
	rm -f hello.elf hello.hex hello32.hex
	rm -f testbench.vvp testbench.vcd
	rm -f testbench_hello.vvp

-include *.d
.PHONY: test clean
```

**testbench_hello.v** (modified from testbench.v)
```
`timescale 1 ns / 1 ps
//`undef VERBOSE_MEM
`define VERBOSE_MEM
//`undef WRITE_VCD
`define WRITE_VCD
`undef MEM8BIT

module testbench_hello;
	reg clk = 1;
	reg resetn = 0;
	wire trap;

	always #5 clk = ~clk;

	initial begin
		repeat (100) @(posedge clk);
		resetn <= 1;
	end

	wire mem_valid;
	wire mem_instr;
	reg mem_ready;
	wire [31:0] mem_addr;
	wire [31:0] mem_wdata;
	wire [3:0] mem_wstrb;
	reg  [31:0] mem_rdata;

	picorv32 #(
		.COMPRESSED_ISA(1)
	) uut (
		.clk         (clk        ),
		.resetn      (resetn     ),
		.trap        (trap       ),
		.mem_valid   (mem_valid  ),
		.mem_instr   (mem_instr  ),
		.mem_ready   (mem_ready  ),
		.mem_addr    (mem_addr   ),
		.mem_wdata   (mem_wdata  ),
		.mem_wstrb   (mem_wstrb  ),
		.mem_rdata   (mem_rdata  )
	);

	localparam MEM_SIZE = 4*1024*1024;
`ifdef MEM8BIT
	reg [7:0] memory [0:MEM_SIZE-1];
	initial $readmemh("hello.hex", memory);
`else
	reg [31:0] memory [0:MEM_SIZE/4-1];
	initial $readmemh("hello32.hex", memory);
`endif

	always @(posedge clk) begin
		mem_ready <= 0;
		if (mem_valid && !mem_ready) begin
			mem_ready <= 1;
			mem_rdata <= 'bx;
			case (1)
				mem_addr < MEM_SIZE: begin
`ifdef MEM8BIT
					if (|mem_wstrb) begin
						if (mem_wstrb[0]) memory[mem_addr + 0] <= mem_wdata[ 7: 0];
						if (mem_wstrb[1]) memory[mem_addr + 1] <= mem_wdata[15: 8];
						if (mem_wstrb[2]) memory[mem_addr + 2] <= mem_wdata[23:16];
						if (mem_wstrb[3]) memory[mem_addr + 3] <= mem_wdata[31:24];
					end else begin
						mem_rdata <= {memory[mem_addr+3], memory[mem_addr+2], memory[mem_addr+1], memory[mem_addr]};
					end
`else
					if (|mem_wstrb) begin
						if (mem_wstrb[0]) memory[mem_addr >> 2][ 7: 0] <= mem_wdata[ 7: 0];
						if (mem_wstrb[1]) memory[mem_addr >> 2][15: 8] <= mem_wdata[15: 8];
						if (mem_wstrb[2]) memory[mem_addr >> 2][23:16] <= mem_wdata[23:16];
						if (mem_wstrb[3]) memory[mem_addr >> 2][31:24] <= mem_wdata[31:24];
					end else begin
						mem_rdata <= memory[mem_addr >> 2];
					end
`endif
				end
				mem_addr == 32'h 1000_0000: begin
					$write("%c", mem_wdata[7:0]);
				end
			endcase
		end
		if (mem_valid && mem_ready) begin
`ifdef VERBOSE_MEM
			if (|mem_wstrb)
				$display("WR: ADDR=%x DATA=%x MASK=%b", mem_addr, mem_wdata, mem_wstrb);
			else
				$display("RD: ADDR=%x DATA=%x%s", mem_addr, mem_rdata, mem_instr ? " INSN" : "");
`endif
			if (^mem_addr === 1'bx ||
					(mem_wstrb[0] && ^mem_wdata[ 7: 0] == 1'bx) ||
					(mem_wstrb[1] && ^mem_wdata[15: 8] == 1'bx) ||
					(mem_wstrb[2] && ^mem_wdata[23:16] == 1'bx) ||
					(mem_wstrb[3] && ^mem_wdata[31:24] == 1'bx)) begin
				$display("CRITICAL UNDEF MEM TRANSACTION");
				$finish;
			end
		end
	end

`ifdef WRITE_VCD
	initial begin
		$dumpfile("testbench.vcd");
		$dumpvars(0, testbench_hello);
	end
`endif

	always @(posedge clk) begin
		if (resetn && trap) begin
			repeat (10) @(posedge clk);
			$display("TRAP");
			$finish;
		end
	end
endmodule

```

I seem to get similar results when running in the firmware folder. But I don't get very far, where as this vcd appears to get many cycles in. Is it possible that there needs to be some kind of clean up code executed in order to exit correctly?

cliffordwolf, 2018-07-30:
Apparently you don't use `start.elf` in your hello world program, nor `syscalls.o`. And you are not including anything else that would provide the same functionality. I don't know what you are expecting to happen if you discard an essential part of the example code without replacing it with anything.

Adding those components back in and everything works as expected:

```
--- Makefile	2018-07-30 14:16:07.580679733 +0200
+++ Makefile	2018-07-30 14:20:18.524726866 +0200
@@ -28,9 +28,10 @@
 	python3 hex8tohex32.py firmware.hex > firmware32.hex
 	rm -f start.tmp firmware.tmp
 
-hello32.hex: hello.elf
+hello32.hex: hello.elf start.elf hex8tohex32.py
+	$(RISCV_TOOLS_PREFIX)objcopy -O verilog start.elf start.tmp
 	$(RISCV_TOOLS_PREFIX)objcopy -O verilog hello.elf hello.tmp
-	cat hello.tmp > hello.hex
+	cat start.tmp hello.tmp > hello.hex
 	python3 hex8tohex32.py hello.hex > hello32.hex
 	rm -f hello.tmp
 
@@ -38,7 +39,7 @@
 	$(CC) $(LDFLAGS) -o $@ $^ -T ../../firmware/riscv.ld $(LDLIBS)
 	chmod -x firmware.elf
 
-hello.elf: hello.o
+hello.elf: hello.o syscalls.o
 	$(CC) $(LDFLAGS) -o $@ $^ -T ../../firmware/riscv.ld $(LDLIBS)
 	chmod -x hello.elf
 
```

closed 2018-07-30 by cliffordwolf

bentomo, 2018-07-30:
Ah, that makes total sense now. I was attempting to strip out as much as possible to figure out what was needed minimally. It makes sense that you need some initialization code for the core. I wasn't kidding when I said noob early. Thanks so much! Working great now!