YosysHQ/picorv32 #77: IceStorm compatibility?

sunnystormy, 2018-08-01:
Heya!

Wanted to reach out and see if this would be compatible with a BlackIce board? I'd love to use the IceStorm tools to get this running!

Thank you!

cliffordwolf, 2018-08-02:
It is compatible witrh IceStorm of course. For example the demo for picosoc is an iCE40 HX8K design that being built with icestorm.

I don't know if there is an already existing picorv32/picosoc design for the BlackIce board, but since the board features a iCE40HX4K (which is just a 8k with a different label) I see no inherent issue with getting picorv32 work on that board.

closed 2018-08-02 by cliffordwolf

ndbroadbent, 2018-11-22:
Hi @cliffordwolf, I'm also interested in running `picorv32` on an iCE40 FPGA. I saw this in the README: "Small (750-2000 LUTs)". So I was wondering if you can build a minimal version that will run on an iCE40HX1K with 1,280 LUTs?

| Chip | LUTs |
|------|------|
| iCE40HX1K | 1,280 |
| iCE40HX4K | 3,520 |
| iCE40HX8K | 7,680 |

(From [this iCE40 datasheet](http://www.latticesemi.com/~/media/LatticeSemi/Documents/DataSheets/iCE/iCE40LPHXFamilyDataSheet.pdf).)

I'm collecting some information to update [my post on /r/FPGA](https://www.reddit.com/r/FPGA/comments/9yutk8/best_100300_fpga_development_board_in_2018/). I've been looking at this affordable [Go Board](https://www.nandland.com/goboard/introduction.html) ($65), which has an iCE40HX1K. I'm just trying to get a sense of what you can run on this Go Board.