YosysHQ/picorv32 #83: Example cxxdemo needs regs 16-31

hutch31, 2018-08-31:
The example code for cxxdemo is compiled using the rv32ic compiler, which I believe uses the full 32 registers of the instruction set.  I had to enable the high regs in the testbench, otherwise it simply gave a TRAP exit.

With attached patch below test case passes.
[testbench.v.patch.txt](https://github.com/cliffordwolf/picorv32/files/2341516/testbench.v.patch.txt)

cliffordwolf, 2018-09-01:
`ENABLE_REGS_16_31 = 1` is already the default value for `ENABLE_REGS_16_31`.

[image]

No need to set it explicitly.

closed 2018-09-01 by cliffordwolf

cliffordwolf, 2018-09-01:
> and didn't realize I hadn't set it back to defaults

No worries. I had assumed it was something like that.