YosysHQ/picorv32 #84: Simulation fails with VCS; VCS does not sign-extend using $signed()

hutch31, 2018-09-01:
I saw this referenced in #33 but closed.  I am using VCS N-2017.12-SP2-1 and traced simulation errors back to sign-extending using $signed, which appears to be the same issue.

After the following code changes I see expected results using VCS.
[picorv32.patch.txt](https://github.com/cliffordwolf/picorv32/files/2341910/picorv32.patch.txt)

cliffordwolf, 2018-09-01:
That's a bug in VCS. If you have support for that, tell Synopsys to fix their f*cking tool. I tried reporting bugs in that piece of crap software before but I didn't get further than "we do not provide support for university licenses". I refuse to masquerade the fact that they are uncapable of fixing trivial bugs in their tool by working around those bugs in my code.

I know it must be frustrating for you, but given the fact that I don't even have access to VCS anymore so I could not even simulate with it after I make changes, I'd rather have it broken in an obvious way that risking tripping obscure VCS bugs whenever I make a trivial change to the design.

If you have Synopsys support, please tell them to get their shit together. Ignoring critical issues like that for many years is just not acceptable.

closed 2018-09-01 by cliffordwolf

hutch31, 2018-09-01:
I understand, but unfortunately do not have the luxury of arguing with Synopsys over who is right...  I will file a STAR but at the moment I'm a really, really small fish to them.

cliffordwolf, 2018-09-01:
jfyi https://twitter.com/oe1cxw/status/1035852371870527488

Re arguing with Synopsys:
(1) :1st_place_medal: for trying.  :laughing:
(2) As I've said, I don't have access to VCS at the moment. But I do have access to Verilog and SystemVerilog standards. If you can produce an MCVE then I can produce the relevant quotes from the standard documents. There really should not be any "arguing" required. They are clearly not correctly implementing IEEE 1364 nor IEEE 1800 here.

hutch31, 2018-09-01:
So there's something deeper about this bug.  I extracted out the failing code into a discrete test case:

```
module signbug;

  reg [31:0] decoded_imm_uj;
  reg [31:0] mem_rdata_latched;

  initial
    begin
      decoded_imm_uj = 32'h0;
      mem_rdata_latched = 32'hF000_0000;

      { decoded_imm_uj[31:20], decoded_imm_uj[10:1], decoded_imm_uj[11], decoded_imm_uj[19:12], decoded_imm_uj[0] } <= $signed({mem_rdata_latched[31:12], 1'b0});

      #1;

      $display("decoded_imm_uj = %x", decoded_imm_uj);
    end

endmodule
```

This code fragment passes in both simulators, but this line fails in VCS when executed in context.  Looks like a good bug.

c60k28, 2018-09-01:
Hello again,
I had been testing your code, in both scenarios I will relate below, have the same result. Please check and comment.
iverilog 10.1.1 Windows:
**C:\iverilog\bin\vvp.exe example.vvp
decoded_imm_uj = fff00700**

VCS L-2016.06-SP2-3 with gcc version 4.7.2 (GCC)
Chronologic VCS simulator copyright 1991-2016
Contains Synopsys proprietary information.
Compiler version L-2016.06-SP2-3_Full64; Runtime version L-2016.06-SP2-3_Full64;  Sep  1 11:00 2018
**decoded_imm_uj = fff00700**
           V C S   S i m u l a t i o n   R e p o r t 
Time: 1
CPU Time:      0.360 seconds;       Data structure size:   0.0Mb
Sat Sep  1 11:01:23 2018
CPU time: .264 seconds to compile + .224 seconds to elab + .344 seconds to link + .492 seconds in simulation

c60k28, 2018-09-01:
Some extra input. Versions 'N-' are experimental in features. Use L or M. I have been testing in M-2017 and L-2016. I may be able to push further analysis, but as I wrote you, N versions had experimental features.

c60k28, 2018-09-01:
Something that caught my attention. All Synopsys Solvnet customers are qualified to open a STAR case, and being VCS a full featured and most purchased products, even most used in other EDA tools, you should have an answer.
Check your installation, compile with full64 flag, change VCS c compiler for GCC, and check again. If that does not fix your issue, download L or M version and fill a case, so VCS R&D can help you. That's the flow.

hutch31, 2018-09-04:
I submitted a test case with the N version and downloaded an M version to try it out.  I believe than the letter simply indicates the release train; as the newest train N is likely to have experimental features.

I also tried with M-2017.03-SP2-10, which had the same result.  We only have 64 bit versions installed, and using system-default gcc didn't have any effect.

I'll update this bug if Synopsys comes back with anything.

hutch31, 2018-09-12:
Synopsys responded to the bug report and confirmed that they see the bug.  The simulation runs correctly when run with -debug_acess+all flag, they are investigating the discrepancy.

c60k28, 2018-09-14:
Quite interesting,
Please share case number.

hutch31, 2018-09-14:
It's case 8001150449, which has been promoted to STAR 9001399377.

Registered SolvNet users may see the full details on this
STAR at:
https://onlinecase.synopsys.com/Home.aspx?starNumber=9001399377

hutch31, 2018-09-21:
Synopsys provided a workaround for this issue, which is to add the "-Xoptflow=off" flag for simulation.  The STAR has gone over to the VCS design team.