YosysHQ/picorv32 #92: Icestudio compatibility (testing on Alhambra)?

drtrigon, 2018-10-14:
I downloaded the picorv32 and picosoc code and tryied to use it in icestudio in order to put it on an Alhambra II board. I was able to integrate picosoc into icestudio and it works so far: Verify reports an error but Build works anyway. Here how it looks like:
[image]
Now I am struggling with how to connect the picosoc pins with my hardware (Alhambra) and how to add code into the cpu? (simple led blink example would do the job)

Can you give me any hints on how to proceed?

Obijuan, 2018-10-18:
Hi!  I have just tested the picorv32 on my Alhambra II board. It is working! :-)

I've uploaded the test and changes to this repo: https://github.com/FPGAwars/Alhambra-II-FPGA/tree/master/examples/picorv32/picosoc

I will test your icestudio block soon, and tell you the changes

Obijuan, 2018-10-18:
I've forgotten to upload the picture of the Picosoc working!
[image]

drtrigon, 2018-10-18:
Very very nice!

I compared your code and the one used by me - only changes are you disabled flash_io2 and 3. My biggest questionmarks are how to embed it into a design and how to upload or add c code (e.g. LED blink example).

Obijuan, 2018-10-18:
I am preparing an example for you, in icestudio.
The firmware should be uploaded in the flash memory using iceprog, outside of icestudio. In future versión we will integrate it into icestudio as well

i've updated the Alhambra-II picosoc repo. The command for sinthesizing and upload the picosoc is:

apio upload -v    (-v is optional. Is for making it verbose)

and for the firmware:

make prog

Obijuan, 2018-10-18:
Ready! You have the example in the repo. It is called demo.ice:

[image]

It takes around 3 minutes to finish. It is better to verify it first (with the version in that repo there are no errors with the verification process)

cliffordwolf, 2018-10-19:
Thanks @Obijuan for looking into this. Looks like you answered the question so I'm now closing this issue.

closed 2018-10-19 by cliffordwolf

drtrigon, 2018-10-21:
I tested it and it works for me too! Thanks again!

I was able to implement equivalents to the Arduino Blink and SerialCallResponseASCII example based on the one given by you.

drtrigon, 2018-10-21:
@Obijuan: Just found https://github.com/FPGAwars/Alhambra-II-FPGA/tree/master/examples/user-test that is very handy!