YosysHQ/picorv32 #95: Firmware update/upload (FPGArduino)

drtrigon, 2018-10-21:
As mentioned in https://github.com/cliffordwolf/picorv32/issues/92; "The firmware should be uploaded in the flash memory using iceprog, outside of icestudio. In future versión we will integrate it into icestudio as well".

What about support for FPGArduino (see http://www.nxlab.fer.hr/fpgarduino/)? That would allow to compile and upload code via Arduino IDE:
[image]

drtrigon, 2018-10-28:
https://github.com/f32c/arduino/issues/32

drtrigon, 2019-03-18:
I re-open this because I still think this would be great idea and as mentioned in https://github.com/f32c/arduino/issues/32#issuecomment-473860377 I was just not able to get it working (even though I know I am really close - I just don't have the time anymore)...

drtrigon, 2019-03-19:
Ok, had another idea. Now we have a bootloader and upload from arduino ide works. However the code does not execute after uploading yet. Looks like a mem mapping issue.

drtrigon, 2019-03-19:
Can someone that knows the picorv32 please help at https://github.com/f32c/arduino/issues/32#issuecomment-474223383 the question is: "..."

cliffordwolf, 2019-03-22:
See answer to #116.

closed 2019-03-22 by cliffordwolf

drtrigon, 2019-03-22:
I was not yet able to get it working but meanwhile I found another way at least to replace the Makefile by the Arduino IDE. Check https://github.com/f32c/arduino/issues/32#issuecomment-475811858 

[AlhambraII_Generic_picorv32.tar.gz](https://github.com/cliffordwolf/picorv32/files/2998748/AlhambraII_Generic_picorv32.tar.gz)

drtrigon, 2019-03-23:
github repo: https://github.com/drtrigon/fpgarduino-icestorm