YosysHQ/picorv32 #97: Bug: global variable declaration and init not possible in one step

drtrigon, 2018-10-22:
During https://github.com/cliffordwolf/picorv32/issues/92 I found that I cannot declare and initialize variables at the same time, like

> int firstSensor = 1;    // first analog sensor
int secondSensor = 2;   // second analog sensor
int thirdSensor = 3;    // digital sensor
int inByte = 0;         // incoming serial byte

it will produce code that does not work. E.g. 2 LEDs being ON even though this never happens in the code used.

In order to make it work I have to split declaration and init like:

> int firstSensor;    // first analog sensor
int secondSensor;   // second analog sensor
int thirdSensor;    // digital sensor
int inByte;         // incoming serial byte
void main()
{
	firstSensor = 1;    // first analog sensor - dummy value
	secondSensor = 2;   // second analog sensor - dummy value
	thirdSensor = 3;    // digital sensor - dummy value
	inByte = 0;         // incoming serial byte
...

cliffordwolf, 2018-10-23:
Which linker script are you using? What `start.s`? Obviously you are not handling the `data`/`sdata` section correctly. But it's impossible to reproduce the issue from the little information you posted. Please post a [MCVE](https://stackoverflow.com/help/mcve).

drtrigon, 2018-10-23:
It's based on https://github.com/FPGAwars/Alhambra-II-FPGA/tree/master/examples/picorv32/picosoc so the same `start.s` as in there. As soon as I have time I will build a MCVE for you.

drtrigon, 2018-10-27:
Use the repo mentioned before and following minimal example:
[firmware.c.txt](https://github.com/cliffordwolf/picorv32/files/2521969/firmware.c.txt)

The LED bar should fill up and the all should flash. But instead on 2 LED turn on even though this never happens in the code used. It seems to be related to the print statement too. Hope this is now a valid MCVE. Greetings.

mmicko, 2018-10-29:
@drtrigon if you could please try to apply latest change to .ld file and confirm if it works on your environment ? Thanks

drtrigon, 2018-10-29:
Works in the test case for me. Also did another test in the originating code were the issue first appeared and this worked too - look like a good fix to me. Thanks!

closed 2018-10-30 by drtrigon

mmicko, 2018-10-30:
Thanks for confirmation.

commit 2f16c469 (2018-10-30): Alignment fix for global symbols fixes #97
commit de479b09 (2018-10-30): Merge pull request #101 from mmicko/master

Alignment fix for global symbols fixes #97