YosysHQ/picorv32 #99: Best process to have multiple implemented using the SB_MAC16 on the iCE40UP5K?

mithro, 2018-10-27:
It is my understanding that the whole multiply instruction inside the picorv32 is implemented by the [`picorv32_pcpi_mul`](https://github.com/cliffordwolf/picorv32/blob/3a6ac16259af8ae6bc6abf8ef1713152fa67f771/picorv32.v#L2127-L2246) module, with the "actual" multiply operation happening on the following line;
https://github.com/cliffordwolf/picorv32/blob/3a6ac16259af8ae6bc6abf8ef1713152fa67f771/picorv32.v#L2301

Currently, I believe Yosys is unable to take this Verilog code and map it to the SB_MAC16 primitives found in the iCE40UP5K part?

Would the best approach to solving this problem be?
 * (a) Adding a custom techmap pass in Yosys to take this multiply operation and map it to the SB_MAC16 primitives.
 * (b) Adding an ifdef around L2301 in picorv32 to replace that with explicit instantiation of the SB_MAC16 block?

It feels like the process of writing (b) would be almost identically to writing the basic techmap pass? Maybe start with (b) and then once someone has the mapping right you can convert it to a techmap pass?

Any idea why nobody has already done this? Am I missing a huge bit of complexity?

rxrbln, 2019-12-15:
AFAICS the iCE40UP5K has a 16x16 bit multiplier, while this 32bit RISCV implementation needs a 32 bit multiplier, ...

roman65536, 2020-12-11:
just came across : https://github.com/Spritetm/hadbadge2019_fpgasoc/tree/master/soc
they use MAC of the ecp5. it gives you an idea how to implement it with iCE40 part.