YosysHQ/picorv32 #106: Clone into submodule path 'riscv-qemu' failed; github.com errno=Connection refused

gojimmypi, 2019-01-15:
After getting this "connection refused" error 3 consecutive times when doing a `sudo make -j$(nproc) build-tools` in exactly the same place:

```
[...snip...]
HEAD is now at c3ad555... Merge pull request #354 from riscv/cherry-pick-351-fix
+ git submodule update --init --reference /var/cache/distfiles/riscv-binutils-gdb.git riscv-binutils
Submodule 'riscv-binutils' (https://github.com/riscv/riscv-binutils-gdb.git) registered for path 'riscv-binutils'
Cloning into 'riscv-binutils'...
Checking connectivity... done.
Submodule path 'riscv-binutils': checked out '0d2fb1b7ff9b94e40352306f1f8857b863c6f787'
+ git submodule update --init --reference /var/cache/distfiles/riscv-binutils-gdb.git riscv-gdb
Submodule 'riscv-gdb' (https://github.com/riscv/riscv-binutils-gdb.git) registered for path 'riscv-gdb'
Cloning into 'riscv-gdb'...
Checking connectivity... done.
Submodule path 'riscv-gdb': checked out '635c14ec58512380f28fc80447a427de9341293d'
+ git submodule update --init --reference /var/cache/distfiles/riscv-dejagnu.git riscv-dejagnu
Submodule 'riscv-dejagnu' (https://github.com/riscv/riscv-dejagnu.git) registered for path 'riscv-dejagnu'
Cloning into 'riscv-dejagnu'...
Checking connectivity... done.
Submodule path 'riscv-dejagnu': checked out '2e99dc08d8e5e16f07627bd52a192906abfa9a5c'
+ git submodule update --init --reference /var/cache/distfiles/riscv-gcc.git riscv-gcc
Submodule 'riscv-gcc' (https://github.com/riscv/riscv-gcc.git) registered for path 'riscv-gcc'
Cloning into 'riscv-gcc'...
Checking connectivity... done.
Submodule path 'riscv-gcc': checked out '489c9e74775b580b9cb57adc80ef3277af5fee11'
+ git submodule update --init --reference /var/cache/distfiles/riscv-glibc.git riscv-glibc
Submodule 'riscv-glibc' (https://github.com/riscv/riscv-glibc.git) registered for path 'riscv-glibc'
Cloning into 'riscv-glibc'...
Checking connectivity... done.
Submodule path 'riscv-glibc': checked out '2f626de717a86be3a1fe39e779f0b179e13ccfbb'
+ git submodule update --init --reference /var/cache/distfiles/riscv-newlib.git riscv-newlib
Submodule 'riscv-newlib' (https://github.com/riscv/riscv-newlib.git) registered for path 'riscv-newlib'
Cloning into 'riscv-newlib'...
Checking connectivity... done.
Submodule path 'riscv-newlib': checked out '320b28ea27c71df7afe62b21a220f77aef9eb88a'
+ git submodule update --init --reference /var/cache/distfiles/riscv-qemu.git riscv-qemu
Submodule 'riscv-qemu' (git://github.com/riscv/riscv-qemu.git) registered for path 'riscv-qemu'
Cloning into 'riscv-qemu'...
fatal: unable to connect to github.com:
github.com[0: 192.30.255.113]: errno=Connection refused
github.com[1: 192.30.255.112]: errno=Connection refused

Clone of 'git://github.com/riscv/riscv-qemu.git' into submodule path 'riscv-qemu' failed
make[1]: *** [build-riscv32i-tools-bh] Error 1
make[1]: Leaving directory `/home/gojimmypi/workspace/picorv32'
make: *** [build-tools] Error 2
```

I found [this similar issue](https://github.com/npm/npm/issues/6285#issuecomment-56640354) from years ago and sure enough, this resolved it for me:

`git config --global url.https://github.com/.insteadOf git://github.com/`

(Thanks @saftacatalinmihai) 

I have no idea why it only fails on the 'riscv-qemu' - I've actually never before encountered this GitHub error.

Not sure if it is worthwhile or desired to update the Makefile (to not use `git://`) to ensure this problem is never encountered by others?

gojimmypi, 2019-02-08:
See also https://github.com/riscv/riscv-qemu/issues/189 

These three settings may be helpful, for instance when building the @tinyfpga [picosoc](https://discourse.tinyfpga.com/t/riscv-example-project-on-tinyfpga-bx/451) with a firewall that blocks `git://` ports:

```
git config --global url.https://git.qemu.org/git/.insteadOf git://git.qemu-project.org/
git config --global url.https://anongit.freedesktop.org/git/.insteadOf git://anongit.freedesktop.org/
git config --global url.https://github.com/riscv.insteadOf git://github.com/riscv
```
Note settings can be undone like this:

`git config --global --unset url.https://github.com/.insteadOf
`

I have some additional details on my blog [here](https://gojimmypi.blogspot.com/2019/01/risc-v-on-fpga-tinyfpga-via-wsl.html) regarding RISC-V on the TinyFPGA in WSL.

cliffordwolf, 2019-03-02:
Looks like this submodule isn't really needed by the build anyways, so I simply removed it now.

closed 2019-03-02 by cliffordwolf