YosysHQ/picorv32 #108: make error in 'script/romload'

balanx, 2019-01-23:
may someone give me some helps ... :(

> make test
/opt/riscv32i/bin/riscv32-unknown-elf-gcc -march=rv32i -Wl,--gc-sections,-Map=firmware.map -o firmware.elf firmware.o syscalls.o start.o -T sections.ld
/opt/riscv32i/lib/gcc/riscv32-unknown-elf/8.2.0/../../../../riscv32-unknown-elf/bin/ld: firmware.elf section `.rom' will not fit in region `rom'
/opt/riscv32i/lib/gcc/riscv32-unknown-elf/8.2.0/../../../../riscv32-unknown-elf/bin/ld: region `rom' overflowed by 10632 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:32: firmware.elf] Error 1

daveshah1, 2019-01-23:
Looks like your program is too big to fit in memory. Either find a way to make it smaller; or increase the memory size in both your design and the linker script.

balanx, 2019-01-31:
I'm a hardware engineer, and very fresh in gnu tools. may you give me some guide for this script/romload example ?

stv0g, 2019-02-11:
Take a look at the at: [`scripts/romload/sections.lds:12`](https://github.com/cliffordwolf/picorv32/blob/5c0e13779210983fb20d908085df4f43037ec132/scripts/romload/sections.ld#L12)

Here you can adapt the length of the ROM. But keep in mind that the size of mapped SPI ROM actually needs to match with this value.

balanx, 2019-02-19:
@stv0g, thanks reply!
I changed the size from 63k to 127k, and it runs OK.

rom(rwx) : ORIGIN = 0x00000100, LENGTH = 127k
ram(rwx) : ORIGIN = 0x00040000, LENGTH = 16k

Again, could you tell me the way to get the size of rom/ram for a .elf file ? what gnu command ... ?
thank you very much!

stv0g, 2019-02-19:
You can inspect the ELF section sizes via

```
readelf -a myelf_file
```

stv0g, 2019-02-19:
But I think the issue is solved. Can you close it please?

balanx, 2019-02-20:
very helpful !
thanks

closed 2019-02-20 by balanx