YosysHQ/picorv32 #112: rv32e support

gsteiert, 2019-02-15:
For the riscv-gnu-toolchain, git rev c3ad555 will not build for the rv32e 
Has anyone been able to build the tools for rv32e, and if so, what version did they use?
What is the status of rv32e support?

gsteiert, 2019-02-17:
I was able to get further adding --with-abi=ilp32e to the configure command:
../configure --with-arch=rv32e --with-abi=ilp32e --prefix=/opt/riscv32e
It still has errors building some of the libraries.
Many of the builds fail with
    Error 1 
    Assembler messages:
    Fatal error: -march=rv32e: first ISA subset must be `i' or `g'

cejpmart, 2019-02-24:
Have you rebuilt everything in a clean build environment?

The error message would suggest that an outdated version of the assembler is still being invoked sometimes.