YosysHQ/picorv32 #115: Error seen while executing make test_verilator

raulbehl, 2019-03-04:
The attached log gives the output of ```make test_verilator```

I have built the riscv-toolchain following the README and can confirm that I am at the mentioned revision of the repo (for riscv-tools). Not sure what went wrong here. 

[verilator.log](https://github.com/cliffordwolf/picorv32/files/2927366/verilator.log)

cliffordwolf, 2019-03-06:
Unfortunately I can not reproduce this with git head of PicoRV32 and git head of Verilator:
[picorv32_test_verilator.log](https://github.com/cliffordwolf/picorv32/files/2933655/picorv32_test_verilator.log)

Is it possible you are using an old version of Verilator?

lucasbrasilino, 2019-03-18:
@raulbehl for me, this issue happens when using an old verilator version. 
Verilator from Ubuntu 16.04 package,version:
```Verilator 3.874 2015-06-06 rev verilator_3_872-20-g0d43051```
Fails just  like you reported.

 Using an updated version, compiled from upstream/sources:
```Verilator 3.922 2018-03-17 rev UNKNOWN_REV```
Test completes successfully.

cliffordwolf, 2019-03-19:
Addressed this issue in f48f5fe.

closed 2019-03-19 by cliffordwolf