YosysHQ/picorv32 #119: Equivalent Gate Count PicoRV32I

APDNath, 2019-04-09:
Hello, 

I tried synthesizing the PicoRV32I core in design compiler using 250 nm LEDA library and got a gate equivalent of 5437 (number of cells in DC area report). I am just wondering if the gate count is right as the number seemed quite low. 
Any help/suggestion/comment would be much appreciated. 

Thanks!

cliffordwolf, 2019-04-17:
> gate equivalent of 5437 (number of cells in DC area report)

What configuration is that? What cell library? I don't have DC, but `yosys -p 'synth -top picorv32; abc -g cmos2; stat' picorv32.v` gives me ~16k gates for the default configuration (that's a cell library with just NAND, NOR, NOT, and FFs). For a pretty minimal RV32E configuration I get ~10k gates. Mapping that minimal config to a cell lib that also includes AOI and OAI gates further reduces the gate count I see to ~7k. Also adding MUX gates further reduces the gate count I see to 5838.

> as the number seemed quite low.

Since I can not reproduce what you are seeing as I neither have your cell library, nor DC, nor know which config you are using, I can only suggest that you run a post-synthesis simulation and see if it behaves as intended.

That being said, the whole point of PicoRV32 is that it's quite small (and easy to integrate).  ;)

closed 2019-04-17 by cliffordwolf

APDNath, 2019-04-24:
Thanks much for the elaborate reply. It is really helpful. :)