YosysHQ/picorv32 #121: Bug: function call

drtrigon, 2019-04-24:
I found that I cannot call some of my functions properly. In order to build an Arduino like structure I use the functions `void setup(void)` and `void loop(void)` and call them from the `main` function. The structure looks like this:
```
void main()
{
	setup();

	while (1) {
		loop();
}
```
It looks like it does not properly return from `setup` as `loop` never gets called. However if I call `loop` from within `setup` it does get called.

The thing I do not understand at all is there are other functions like `print` that do work properly (so they return and code after them does get called).

I can derive a full MCVE as soon as I have more time, but basically it is a bit modified version of https://github.com/cliffordwolf/picorv32/issues/92 to work as blink example.

yanghao, 2019-04-24:
Very likely your stack is not setting up properly, or your BSS is not getting initialized. I have a customized complex piece of SW running on picorv32 perfectly fine (after enabling unaligned access).

drtrigon, 2019-04-24:
So I guess that's related to:
* https://github.com/FPGAwars/Alhambra-II-FPGA/blob/master/examples/picorv32/picosoc/start.s
* https://github.com/mmicko/picorv32/blob/master/picosoc/sections.lds

I just realized that there are more recent files on https://github.com/cliffordwolf/picorv32/tree/master/picosoc - I have to merge them in and test again. Will report my findings.

yanghao, 2019-04-24:
PS: I was not using the picorv32 linker scripts ... any linker scripts and start up code as long as it does all the necessary things should work (setup stack, zero bss). You can enable the register debug option and do a simulation w/ your binary and check core registers and PC pointers etc.

drtrigon, 2019-04-25:
@yanghao: I am just a physicist as such lack the proper background, so any hint and explanation is very welcome, thanks! I am ignorant because I never took lectures regarding this fancy stuff. ;)

So meanwhile I was able to make it working by merging in the latest code from here, this is the diff:

```
--- a/Alhambra_picorv32-picosoc/Blink_test-recent-picorv32/Makefile                                                                                                                                                                                                            
+++ b/Alhambra_picorv32-picosoc/Blink_test-recent-picorv32/Makefile                                                                                                                                                                                                            
@@ -9,16 +9,11 @@ prog: $(FIRMWARE).bin                                                                                                                                                                                                                                        
 #      iceprog -o 1M $(FIRMWARE).bin                                                                                                                                                                                                                                          
        $(ICESTORM)/iceprog -o 1M $(FIRMWARE).bin                                                                                                                                                                                                                              
                                                                                                                                                                                                                                                                               
-$(FIRMWARE).elf: sections.lds start.s firmware.c                                                                                                                                                                                                                              
-       $(RISCV)/riscv32-unknown-elf-gcc -DHX8KDEMO -march=rv32imc -Wl,-Bstatic,-T,sections.lds,--strip-debug -ffreestanding -nostdlib -o $(FIRMWARE).elf start.s firmware.c                                                                                              
+hx8kdemo_sections.lds: sections.lds
+       $(RISCV)/riscv32-unknown-elf-cpp -P -DHX8KDEMO -o $@ $^
+
+$(FIRMWARE).elf: hx8kdemo_sections.lds start.s firmware.c
+       $(RISCV)/riscv32-unknown-elf-gcc -DHX8KDEMO -march=rv32imc -Wl,-Bstatic,-T,hx8kdemo_sections.lds,--strip-debug -ffreestanding -nostdlib -o $(FIRMWARE).elf start.s firmware.c
 
 $(FIRMWARE).hex: $(FIRMWARE).elf
        $(RISCV)/riscv32-unknown-elf-objcopy -O verilog $(FIRMWARE).elf $(FIRMWARE).hex

--- a/Alhambra_picorv32-picosoc/Blink_test-recent-picorv32/firmware.c
+++ b/Alhambra_picorv32-picosoc/Blink_test-recent-picorv32/firmware.c
@@ -20,7 +20,11 @@
 #include <stdint.h>
 #include <stdbool.h>
 
-#if !defined(ICEBREAKER) && !defined(HX8KDEMO)
+#ifdef ICEBREAKER
+#  define MEM_TOTAL 0x20000 /* 128 KB */
+#elif HX8KDEMO
+#  define MEM_TOTAL 0x200 /* 2 KB */
+#else
 #  error "Set -DICEBREAKER or -DHX8KDEMO when compiling firmware.c"
 #endif
 
@@ -182,11 +187,21 @@ void print_hex(uint32_t v, int digits)
 
 void print_dec(uint32_t v)
 {
-       if (v >= 100) {
-               print(">=100");
+       if (v >= 1000) {
+               print(">=1000");
                return;
        }
 
+       if      (v >= 900) { putchar('9'); v -= 900; }
+       else if (v >= 800) { putchar('8'); v -= 800; }
+       else if (v >= 700) { putchar('7'); v -= 700; }
+       else if (v >= 600) { putchar('6'); v -= 600; }
+       else if (v >= 500) { putchar('5'); v -= 500; }
+       else if (v >= 400) { putchar('4'); v -= 400; }
+       else if (v >= 300) { putchar('3'); v -= 300; }
+       else if (v >= 200) { putchar('2'); v -= 200; }
+       else if (v >= 100) { putchar('1'); v -= 100; }
+
        if      (v >= 90) { putchar('9'); v -= 90; }
        else if (v >= 80) { putchar('8'); v -= 80; }
        else if (v >= 70) { putchar('7'); v -= 70; }

--- a/Alhambra_picorv32-picosoc/Blink_test-recent-picorv32/picosoc.v
+++ b/Alhambra_picorv32-picosoc/Blink_test-recent-picorv32/picosoc.v
@@ -25,6 +25,14 @@
 `define PICORV32_REGS picosoc_regs
 `endif
 
+`ifndef PICOSOC_MEM
+`define PICOSOC_MEM picosoc_mem
[... 25 lines cut ...]
+#  define MEM_TOTAL 0x20000 /* 128 KB */
+#elif HX8KDEMO
+#  define MEM_TOTAL 0x200 /* 2 KB */
+#else
+#  error "Set -DICEBREAKER or -DHX8KDEMO when compiling firmware.c"
+#endif
+
 MEMORY
 {
     FLASH (rx)      : ORIGIN = 0x00100000, LENGTH = 0x400000 /* entire flash, 4 MiB */
-    RAM (xrw)       : ORIGIN = 0x00000000, LENGTH = 0x000400 /* 1 KB */
+    RAM (xrw)       : ORIGIN = 0x00000000, LENGTH = MEM_TOTAL
 }
 
 SECTIONS {
@@ -15,6 +23,7 @@ SECTIONS {
         *(.rodata*)        /* .rodata* sections (constants, strings, etc.) */
         *(.srodata)        /* .rodata sections (constants, strings, etc.) */
         *(.srodata*)       /* .rodata* sections (constants, strings, etc.) */
+        . = ALIGN(4);
         _etext = .;        /* define a global symbol at end of code */
         _sidata = _etext;  /* This is used by the startup in order to initialize the .data secion */
     } >FLASH
```
...even though I did not verify it (just a guess), I'm quite sure it was the linker script, namely the missing `ALIGN(4)`. This cause already such kind of strange issues some time ago: #97. Somehow I must have missed the fix in this example - sorry for the confusion!

closed 2019-04-25 by drtrigon