YosysHQ/picorv32 #124: How to handle section .eh_frame in linker script

drtrigon, 2019-05-02:
I do get this when trying to link:

```
riscv32-unknown-elf/bin/ld: section .eh_frame LMA [0000000000101264,00000000001019bf] overlaps section .data LMA [0000000000101264,0000000000101277]
```

in the [linker script](https://github.com/cliffordwolf/picorv32/blob/master/picosoc/sections.lds) is no entry for the section `.eh_frame`. If I add `KEEP (*(.eh_frame))` e.g. it works, but I don't know whether that is correct and where to add exactly. What do I need to do?

drtrigon, 2019-05-04:
Actually it also works without using this section if using `riscv32-unknown-elf-g++` and removing `-nostdlib`.

But the new code does not fit in size anymore:
```
riscv32-unknown-elf/bin/ld: /tmp/arduino_build_459262/DigitalReadSerial.ino.elf section `.data' will not fit in region `RAM'
riscv32-unknown-elf/bin/ld: region `RAM' overflowed by 9528 bytes
```

drtrigon, 2019-05-04:
How much RAM is safe to assign in the linker script? The Alhambra II should have at least 80kbits RAM available but the linker script uses only 2 KB. Can I increase the value? How much is safe?

drtrigon, 2019-05-04:
According to https://lists.freebsd.org/pipermail/freebsd-hackers/2011-October/036652.html using `-fomit-frame-pointer -fno-exceptions -fno-asynchronous-unwind-tables -fno-unwind-tables` suppresses generation of `.eh_frame` section and solves this issue for me.

closed 2019-05-04 by drtrigon

nbelanger99, 2020-04-19:
> According to https://lists.freebsd.org/pipermail/freebsd-hackers/2011-October/036652.html using `-fomit-frame-pointer -fno-exceptions -fno-asynchronous-unwind-tables -fno-unwind-tables` suppresses generation of `.eh_frame` section and solves this issue for me.

I'm having the same problem, using the same toolchain, but using those flags didn't get rid of the eh_frames section for me.