YosysHQ/picorv32 #126: Performance

drtrigon, 2019-06-18:
Naiive question: is it possible that the following code needs 256 cycles, every further iteration (++us) are additional 197 cycles?

```
unsigned int us = 1;
__asm__ __volatile__ (
	  "1: addi %0,%0,-1" "\n\t"
	  "bnez %0,1b" : "=r" (us) : "r" (us)
	);
```

I timed it using 

```
  __asm__ volatile ("rdcycle %0" : "=r"(a));
```
with `a` beeing a `uint32_t`.

This seems like a lot just for an add and a compare/branch also considering https://github.com/cliffordwolf/picorv32#cycles-per-instruction-performance.

The reason why I'm asking is I try to implement and equivalent to the Arduino `delayMicroseconds` and this delay code takes between 16-21 us on an Alhambra II board running at 12 MHz.

daveshah1, 2019-06-18:
Are you running the code out of SPI flash by any chance?

drtrigon, 2019-06-19:
Yes I am. Is there a way to speed this up? Cache or anything else?

What setup is needed to achive the numbers given in https://github.com/cliffordwolf/picorv32#cycles-per-instruction-performance ?

cliffordwolf, 2019-06-20:
> Cache or anything else?

Yes, you can create a cache. Or you can just copy all performance-critical code to RAM, or execute from a ROM. Whatever works for you. These are decisions you have to make about the system you are building and have nothing to do with PicoRV32 itself, as PicoRV32 is just the processor core.

> What setup is needed to achieve the numbers given in [..]

Just a fast RAM. See https://github.com/cliffordwolf/picorv32/blob/master/dhrystone/testbench.v for the setup.

closed 2019-06-20 by cliffordwolf

drtrigon, 2021-11-04:
Sorry for comming back to this late, but my time is restricted... ;)

The project is: https://github.com/drtrigon/fpgarduino-icestorm

I still would like to make this work. As hardware I use an Alhambra board. As development tool icestudio. I can guess of 2 possible ways to speed this up:
1. use a cache as mentioned; can you give me some hints on how to implement this in icestudio?
2. use a PLL to generate a faster clock; here I would need some hints on how to implement and use a faster clock?