YosysHQ/picorv32 #130: Consider providing Prerequisites section for PicoSoC

kuzminrobin, 2019-08-09:
(I never dealt with Verilog before, I'm a software dev)
At https://github.com/cliffordwolf/picorv32/tree/master/picosoc (`README.md`) I see
> `make hx8ksim`
> `make hx8kprog`

I get:
> `yosys-config: not found`
> `iverilog: not found`
> `yosys: Command not found`

Consider providing the Prerequisites section to the `README.md`, something like
**Prerequisites.**
`sudo apt-get install ..yosys.. ..verilog..`

dangerggg, 2019-10-28:
I installed yosys from apt-get, but in the cell_sim.v file I cannot find the SPRAM_256KA...

cjhonlyone, 2020-04-02:
> I installed yosys from apt-get, but in the cell_sim.v file I cannot find the SPRAM_256KA...

You should install yosys-dev also for yosys--config.