YosysHQ/picorv32 #132: iverilog simulate error: jump to wrong address, and decode instruction at wrong address

XIVN1987, 2019-09-07:
my c code:
```c
void entry(void)
{
    extern unsigned int _bss;
    extern unsigned int _ebss;

    for (unsigned int *dst = &_bss; dst < &_ebss; dst++)
    *dst = 0;

    main();
}
```

compiled by riscv-none-embed-gcc, and generate disassembly code:
```assemble
00000000 <entry>:
       0:	1101                	addi	sp,sp,-32
       2:	ce06                	sw	ra,28(sp)
       4:	cc22                	sw	s0,24(sp)
       6:	1000                	addi	s0,sp,32
       8:	6789                	lui	a5,0x2
       a:	fd478793          	addi	a5,a5,-44 # 1fd4 <_etext>
       e:	fef42623          	sw	a5,-20(s0)
      12:	a811                	j	26 <entry+0x26>
      14:	fec42783          	lw	a5,-20(s0)
      18:	0007a023          	sw	zero,0(a5)
      1c:	fec42783          	lw	a5,-20(s0)
      20:	0791                	addi	a5,a5,4
      22:	fef42623          	sw	a5,-20(s0)
      26:	fec42703          	lw	a4,-20(s0)
      2a:	6789                	lui	a5,0x2
      2c:	ff478793          	addi	a5,a5,-12 # 1ff4 <_ebss>
      30:	fef762e3          	bltu	a4,a5,14 <entry+0x14>
      34:	068010ef          	jal	ra,109c <main>
      38:	40f2                	lw	ra,28(sp)
      3a:	4462                	lw	s0,24(sp)
      3c:	6105                	addi	sp,sp,32
      3e:	8082                	ret
```

iverilog debug message:
```
------ reset = 1
------ reset = 0
ST_RD:   2 0x00004000
-- 60000
DECODE: 0x00000000     0xxxxx UNKNOWN
LD_RS1:  x 0xxxxxxxxx
LD_RS2:  x 0xxxxxxxxx
-- 120000
DECODE: 0x00000004     0x1101 addi
LD_RS1:  2 0x00004000
ST_RD:   2 0x00003fe0
-- 150000
LD_RS1:  2 0x00003fe0
LD_RS2:  1 0xxxxxxxxx
DECODE: 0x00000006     0xce06 sw
-- 200000
DECODE: 0x00000008     0xcc22 sw
LD_RS1:  2 0x00003fe0
LD_RS2:  8 0xxxxxxxxx
-- 250000
LD_RS1:  2 0x00003fe0
DECODE: 0x0000000a     0x1000 addi
ST_RD:   8 0x00004000
-- 280000
DECODE: 0x0000000c     0x6789 lui
ST_RD:  15 0x00002000
-- 320000
DECODE: 0x0000000e 0xfd478793 addi
LD_RS1: 15 0x00002000
ST_RD:  15 0x00001fd4
-- 360000
DECODE: 0x00000012 0xfef42623 sw
LD_RS1:  8 0x00004000
LD_RS2: 15 0x00001fd4
-- 410000
ST_RD:   0 0x00000018, BRANCH 0x0000002a
DECODE: 0x00000016     0xa811 jal
-- 440000
DECODE: 0x0000002a     0xfec4 UNKNOWN
```

there is two error:
1. the disassembly code is "j	26"，but iverilog say "BRANCH 0x0000002a", does the picorv32 jump to wrong address?
2. the code at address 2a is "6789", but iverilog say "DECODE: 0x0000002a     0xfec4 UNKNOWN", it looks like that 0xfec4 is the data at address 28, does the picorv32 decode instruction at wrong address?

thanks a lot

XIVN1987, 2019-09-07:
I opened 'DEBUGASM' debug option, and then notice that the ready_o signal of memory rised before rdata_o signal, so picorv32 fetched wrong instruction data for address 0

I have fixed this problem, so picorv32 can get right instruction data for address 0, but the instruction at address 0 excutes two times:

```
------ reset = 1
------ reset = 0
ST_RD:   2 0x00004000
-- 70000
LD_RS1:  2 0x00004000
DECODE: 0x00000000     0x1101 addi
debugasm 00000000 00001101     addi
ST_RD:   2 0x00003fe0
-- 100000
debugasm 00000002 0000ce06       sw
DECODE: 0x00000002     0xce06 sw
LD_RS1:  2 0x00003fe0
LD_RS2:  1 0xxxxxxxxx
-- 150000
LD_RS1:  2 0x00003fe0
DECODE: 0x00000004     0x1101 addi
debugasm 00000004 00001101     addi
ST_RD:   2 0x00003fc0
-- 180000
debugasm 00000006 0000ce06       sw
DECODE: 0x00000006     0xce06 sw
LD_RS1:  2 0x00003fc0
LD_RS2:  1 0xxxxxxxxx
-- 230000
LD_RS1:  2 0x00003fc0
LD_RS2:  8 0xxxxxxxxx
DECODE: 0x00000008     0xcc22 sw
debugasm 00000008 0000cc22       sw
-- 280000
debugasm 0000000a 00001000     addi
DECODE: 0x0000000a     0x1000 addi
LD_RS1:  2 0x00003fc0
ST_RD:   8 0x00003fe0
-- 310000
DECODE: 0x0000000c     0x6789 lui
debugasm 0000000c 00006789      lui
ST_RD:  15 0x00002000
-- 350000
LD_RS1: 15 0x00002000
DECODE: 0x0000000e 0xfd478793 addi
debugasm 0000000e fd478793     addi
ST_RD:  15 0x00001fd4
-- 390000
LD_RS1:  8 0x00003fe0
LD_RS2: 15 0x00001fd4
DECODE: 0x00000012 0xfef42623 sw
debugasm 00000012 fef42623       sw
-- 440000
debugasm 00000016 0000a811      jal
DECODE: 0x00000016     0xa811 jal
ST_RD:   0 0x00000018, BRANCH 0x0000002a
-- 470000
LD_RS1:  0 0x00000000
LD_RS2:  0 0x00000000
DECODE: 0x0000002a     0xfec4 UNKNOWN
```

cliffordwolf, 2019-09-07:
Since you do not provide sufficient information to reproduce what you are seeing there is nothing I can do to help.

XIVN1987, 2019-09-07:
[RISCV-pico.zip](https://github.com/cliffordwolf/picorv32/files/3586459/RISCV-pico.zip)

this is my files, go to RISCV-pico/sim folder and excute make, then you can reproduce my problem.

thanks a lot.

XIVN1987, 2019-09-07:
I found the problem, my mistake.

I set the memory ready_o signal always high, so picorv32 fetched wrong instruction.

closed 2019-09-07 by XIVN1987