YosysHQ/picorv32 #142: picorv32_pcpi_div register reset

smurray798, 2019-11-14:
None of the registers in the picorv32_pcpi_div module have a reset value, giving rise to potentially undefined behavior. My own post-translate testbench simulations (but not behavioral simulations) of the CPU have resulted in failures after attempting to execute a div or rem instruction because the outputs of picorv32_pcpi_div are uninitialized registers. The undefined values of the registers then propagate into the CPU core, corrupting the signals pcpi_int_* first, and then others later, which prevents the CPU from continuing with the simulation.

Fortunately, there is an easy fix: simply add a reset statement for all the registers in the module. My simulations show that initializing all registers in the module to '0' upon reset removes the undefined registers problem.

Some notes for those who care about why this problem occurred:

This issue does not seem to affect FPGAs. My own testing with a Xilinx Spartan VI did not demonstrate the problem either in simulation (both behavioral and post-translate) or in physical implementation. I didn't dig into the inferred LUTs or gates, but I suspect that the FPGA hardware resets/initializes its flip flops to 0 by default, which would avoid the problem. This is likely the case with all FPGA-based designs, which may be why the issue hasn't appeared for other users yet.

However, my implementation of the CPU is for microchips, not FPGAs. The logic gate standard cell library at my disposal includes flip flops that have no set or reset signal, hence, their output is undefined until their first clock pulse. picorv32_pcpi_div does not specify any reset statements for the module's registers, so the synthesis tool (Cadence's Genus RTL compiler) infers them as flip flops without reset signals. Furthermore, the synthesis tool includes intelligent clock gating optimization software that automatically inserts clock gates in front of modules so that the module is only clocked when it needs to be. In the case of the picorv32_pcpi_div module, that means it doesn't receive a clock signal until a div or rem instruction is called, so at the time of the first clock cycle in the instruction, the internal registers are still undefined, and the issue is manifested.

The issue does not affect behavioral simulations because there is no clock gating written into the Verilog, so the registers are initialized with the first clock pulse. Since the CPU uses the first clock pulse to fetch the first instruction from memory, the first clock pulse cannot coincide with the _execution_ of a rem or div instruction, and therefore cannot demonstrate the problem.

cliffordwolf, 2019-11-18:
> None of the registers in the picorv32_pcpi_div module have a reset value

Are you sure you are using an up-to-date version of the code? All of picorv32 uses synchronous resets and `instr_div`, `instr_divu`, `instr_rem`, `instr_remu`, `pcpi_wait`, `pcpi_wait_q`, `pcpi_ready`, `pcpi_wr`, and `running` in `picorv32_pcpi_div` are all set to zero on reset.

yanghao, 2019-11-18:
Normally start up assembly code when setting up the C environment would zero all x1 ~ x31 (except x2 for stack). If you use any commercial IP (e.g. ARM) they are also not setting core registers to zero and if you don't do this in your startup assembly code you might see a lot of X in the simulation.

My own firmware always does this so I guess this is the reason I don't see any problem on picorv32. My suggestion is make sure you do this in your firmware startup.

And this is why having random default value is much better than having X in the simulation ... X makes sure everything just fails but there are cases you really don't care the initial value and in physical implementation you will never have X ...

smurray798, 2019-11-18:
@cliffordwolf : Looking specifically at the four reg signals `instr_div`, `instr_divu`, `instr_rem`, and `instr_remu`, note that they have no _asynchronous_ reset value, though they are initialized to `0` with every `clk` rising edge (unless the subsequent code in the process overrides them to `1`). With no asynchronous reset value, the `instr_*` signals are realized as flip flops without set or reset inputs. The four `instr_*` signals are OR'ed together to get `instr_any_div_rem`, which sets the value of `pcpi_wait` on every `clk` rising edge. `start` depends on `pcpi_wait`, `running` depends on `start`, `pcpi_ready` depends on `running`, and `pcpi_ready` determines weather or not those first four `instr_*` signals get updated to `1` or not. In behavioral simulations, this closed loop is not an issue, since `clk` is not gated, and therefore the loop has an initial value. However, when the synthesis tool has the ability to insert clock gates within modules in an effort to save power, it does not recognize that you want the `instr_*` signals to have any sort of reset value because it only knows that they should default to `0` on every `clk` pulse unless overridden. Therefore, the synthesis tool automatically gates `clk` so that it only runs when it is needed. Incidentally, the four `instr_*` signals become dependencies of the clock gating logic, and since they are initially undefined, the clock gating logic is too, creating an inescapable loop where `clk` can never have a _defined_ rising edge. Under such circumstances, `pcpi_wait` is always undefined, so as soon as the CPU tries to access any PCPI interface (including the multiplier), the undefined `pcpi_div_wait` signal corrupts the rest of the CPU.

@yanghao : My apologies, I believe there is a slight miscommunication. When I mentioned "registers" in my previous post, I was referring to the generic Verilog `reg` type, not a CPU general purpose register. To be clear, the issue does not deal with x1 ~ x31.

cliffordwolf, 2019-11-19:
> Looking specifically at the four reg signals instr_div, instr_divu, instr_rem, and instr_remu, note that they have no asynchronous reset value

As I wrote In my reply above:

> All of picorv32 uses synchronous resets

Furthermore:

> However, when the synthesis tool has the ability to insert clock gates within modules in an effort to save power, it does not recognize that you want the instr_* signals to have any sort of reset value because it only knows that they should default to 0 on every clk pulse unless overridden.

This sound to me like your problem is with that tool, not with the core, or with the fact that you seem to not properly reset the core. The tool should be able to figure out that it can't gate the clock in a cycle where reset is active. Since I don't have access to that tool I can't really do anything here.

closed 2019-11-19 by cliffordwolf