YosysHQ/picorv32 #147: port linux and picorv32 on fpga

winner96, 2020-01-21:
Hi. I'm newbie on these tasks.

I wants to port picorv32 on my fpga board. (KC705 or ZCU102)

I tried to find various guides, but it was hard to find.

which looks better to port picorv32 using axi-module or picosoc on fpga?

is it possible to add uart IP beside picorv32 and see linux displays on monitor?

thanks :)

jeremyherbert, 2020-01-21:
Linux needs an MMU (which I'm pretty sure picorv32 does not have) so it won't be possible to boot linux on it. I think this project is more suitable for applications where you would normally find a microcontroller.

If you are looking to boot Linux on an FPGA using RISCV then perhaps you could look at the litex project which uses VexRiscV to boot linux on ECP5 FPGAs.

winner96, 2020-01-21:
Thanks for comments.

I have some more question as newbie.

https://unix.stackexchange.com/questions/190350/mmu-less-kernel

is it possible to build linux kernel with no mmu support. and If it succeed, still possible to port on picorv32?

let me first look into VexRiscV. Thanks :)

jeremyherbert, 2020-01-21:
You can build MMU-less linux (also known as uclinux) but it is very tricky to use because there is no virtual memory support. Errors that would lead to your program crashing on a PC will take down the entire kernel. 

I think the "pico" in picorv32 is enough evidence that this processor was not designed to be used with linux.

winner96, 2020-01-22:
Sincerely thank you for your help.

But I still do not understand why it makes error. If uclinux does not support virtual memory and picorv32 have no MMU, what makes PC to crash..

sorry for bothering u

bittenbybytes, 2020-01-25:
The MMU protects memory regions against non-priviledged access(either accidental or malicious) on a hardware level. Without it any faulty code can corrupt sensitive kernel memory. As jeremyherbert hinted at: this core is optimized to be very small for low resource use and not intended for projects requiring large operating systems(maybe something like FreeRTOS at most). If your project absolutely requires Linux and your hardware has the required logic resource and memory, this low end version of the Risc-V is not the best choice.

winner96, 2020-01-27:
Thanks for reply  :)
it helps me a lot.