YosysHQ/picorv32 #155: Request for an example that tiggers an interrupt from HDL.

mkvenkit, 2020-04-01:
`firmware/` demonstrates IRQs but it is set up using assembly, and I am not sure if it shows how to trigger an IRQ from an external source like a push button. 

The documentation says:

```
An interrupt can be triggered by asserting the corresponding bit in the irq input of the core.
```

So, it seems that for instance, if I set the `irq_5` input to `picosoc' I should trigger an interrupt? But that seems to hang the processor.

Please provide an example or hints on how to implement a "button press" interrupt handler since this is a common situation. 

Thanks!

mkvenkit, 2020-04-01:
I was able to figure this out. Closing.

closed 2020-04-01 by mkvenkit

pablogs9, 2020-04-19:
Is it possible to share this code? Thanks!

mkvenkit, 2020-04-19:
@pablogs9  - I've pushed the code here:

https://github.com/electronut/learn-fpga/tree/master/ice40up5k/picosoc_gpio

I've put in code so that the SoC keeps periodically generating an interrupt, and it prints to UART.

The files you need to check are:

```
firmware_lice40up.c
lice40up.v
start.S
```

I am going through a learning curve myself, so I am sure I have missed things. Good luck!

pablogs9, 2020-04-19:
Thanks very much!!