YosysHQ/picorv32 #159: simulation error

bilkarpooja, 2020-04-25:
hello this is pooja since from a couple of weeks i m working on picorv32 code bt i m facing one error after simulating the picosoc folder of the picorv32 ,plz  help me to debud this error. the error snap is below
[image]

bilkarpooja, 2020-04-25:
[image]
here i m getting xxxx values for cpu rs1 and rs2 wats the problem in it

cliffordwolf, 2020-04-29:
In the screenshot you provided it complains about not being able to open the file `firmwari.hex`. That's not a filename ever used in the official version of picosoc, so obviously you are using some local modifications. But since you have not provided those local modifications there is no way for someone else to replicate the issue. Thus there is nothing anyone can do to help you sort out this issue.

closed 2020-04-29 by cliffordwolf