YosysHQ/picorv32 #164: Beginner question: C.JR not branching

elchanan18, 2020-05-18:
After executing the  C.JR at cycle 12, with the contents of rs1=8, shouldn't the next rvfi_pc_rdata at cycle 15 be 8, and not just ignore the branch and increment to 4?

[image]

elchanan18, 2020-05-19:
Never mind. :-)  I found my decode error.

Abhishek7799-max, 2021-07-07:
Hey Bro, I am also a beginner here and I want to know how can I run only the picorv core(i.e. basic instructions from the instruction set). I am using the Windows platform and Xilinx ISE. I don't know how to check the functionality of the core. I downloaded the project and have put all the  Verilog source codes and test benches in ISE. It successfully synthesizes and runs but, of course, gives no particular answers as I don't know how to do it!!!