YosysHQ/picorv32 #170: Run "arbitrary" C code on picorv32

MakisChristou, 2020-07-16:
Sorry if this is a dumb question, but i can't seem to figure it out on my own. I am trying to use simple RISC-V implementations and run some C benchmarks on them. The problem is that I can't seem to figure out how to do this. As far as I can tell, it's not just a simple `riscv32-unknown-elf-gcc main.c -o main.elf` and be done with it. I looked at the makefile and what it does in order to compile C code into binaries and simulate them in picorv32 but I didn't understand much. It seems to need(?) some assembly code first and then jump to the individual C programs/functions(?). Is there a "simple" way to compile any reasonably sized C code and run it baremetal? What are the high level steps that are involved in this process? Any pointers or direction would be greatly appreciated!

johnwinans, 2020-07-16:
It can be CLOSE to riscv32-unknown-elf-gcc main.c -o main.elf.  I think the following class lectures might be of interest to you:

https://youtu.be/tycK1znD8Bw (be patient, the exact stuff you want starts around 46:00 into this video)
https://youtu.be/MTroisf0FsM (this continues to discuss the details all the way up into C++ with STL.)

The compiler & tools used in that course are from here: https://github.com/riscv/riscv-gnu-toolchain

These are part of a playlist I created for some covid-19 lectures that include RISC-V topics for a course where the students wrote simulators to execute RV32I code. The course notes and stuff are (at the moment, it looks like NIU is tinkering with their servers. My faculty pages appear to be a bit jumbled at the moment. But the good stuff is) all here:

http://faculty.cs.niu.edu/~winans/CS463/#RISCV

See the riscv-tests.zip link for the source of Makefiles, example startup files, and how to use them featured int he videos.

Let me know if this helps.

-John

MakisChristou, 2020-07-22:
Thank you for the info and the material. Although I haven't finished the lectures yet, I managed to use the script in `~/picorv32/scripts/cxxdemo` which seems to be working "out of the box" with C as well as C++ code using the standard libraries. The thing is, my goal is to run mibench on the picorv32 and that code uses the `fopen` function which when running the code (using the makefile in said directory) I get the following output: `Unimplemented system call called!
TRAP` and then the code exits. What has to be done (in a high level) in order to prevent this? Should I store the files in the CPUs memory beforehand somehow? Again, thanks for the answer and material!

mguthaus, 2020-07-22:
Interesting to see MiBench still being used. (It was my MS project many years ago!)

MakisChristou, 2020-07-29:
What would be the easiest way to get the cxxdemo working with verilator? I tried modifying the testbench given on in the cxxfolder but with no luck. When I run the verilator model I get a CRITICAL UNDEF MEM TRANSACTION. https://pastebin.com/sz4du70x Any ideas? Thanks in advance.