YosysHQ/picorv32 #178: Help me to locate 1Read 1Write picorv_register module

zeeshanrafique23, 2020-10-22:
I hear 20th-oct talk of Open-RAM, they said that picorv32 is using a register file that has a 1R and 1W port. I searched in the picosoc.v and in the whole repo but found nothing like that anyone who can search it and send the link in a comment.

mguthaus, 2020-10-22:
https://github.com/cliffordwolf/picorv32/blob/master/picosoc/picosoc.v

Lines 222-257.

mguthaus, 2020-10-22:
Also, this option:
https://github.com/cliffordwolf/picorv32#enable_regs_dualport-default--1

zeeshanrafique23, 2020-10-22:
Hey thanks, Matthew I am just listening to you...

zeeshanrafique23, 2020-10-22:
If we have one read port should they stall for another register read operation or they are using any other technique?

mguthaus, 2020-10-23:
I would assume so. The code is all there so you can check.

zeeshanrafique23, 2020-10-23:
Thanks Prof.  Matt

closed 2020-10-24 by zeeshanrafique23