YosysHQ/picorv32 #179: is dedicated irq stack needed?

jallennk, 2020-10-23:
The example start.S switches to a dedicated stack for the interrupt handler.  Is this strictly necessary?  I mean can I just leave the stack pointer alone so that whatever the current stack is used for the interrupt handler?  I think it should be OK as long as GCC does not generate code that accesses memory below the current stack pointer.  I assume this is the case, but does anybody know for sure?

smurray798, 2020-12-30:
Using a dedicated interrupt handler stack is not strictly necessary, but it's a perfectly valid option.

However, you should be careful: an interrupt will almost certainly modify some of your CPU registers. If you don't save the pre-IRQ state of the registers, then your main code will get messed up when the IRQ returns because the main code would not anticipate a sudden change in the registers. This relates back to the stack because in the example start.S, the first thing the interrupt handler does is dump all the registers into the IRQ stack. It then executes the desired interrupt code, happily and freely modifying the registers, and then puts the old values of the registers back just like it found them as it returns control to the main code.

In your case, you don't need to have a dedicated IRQ stack if you don't want to. One option is to simply push all the CPU registers on the top of the main code's stack immediately when the interrupt triggers, and then pop them back off at the end of the interrupt. This option is nice because you don't need to keep adjusting the size of your IRQ stack depending on how much stack memory you think you'll need for it. As long as you don't have a stack overflow on your main stack (which the dedicated IRQ stack would not prevent, anyway), you'll be fine.

>  I mean can I just leave the stack pointer alone so that whatever the current stack is used for the interrupt handler?

Be careful here, you _don't_ want to use any of the data on top of the main code's stack. If you modify any of the data there, you'll corrupt whatever the main code was doing before the interrupt. There is nothing useful to be gleaned from this data either, as you won't know at what point the interrupt triggered during the main code's execution. You can push _more_ data onto the stack (as long as you pop it off at the end of the interrupt), but you can't modify _existing_ stack data.

jyukumite, 2021-01-05:
This might give someone a leg up - caveat emptor etc. - it's simple, written in c (including the init), doesn't do any cpp init.

Instantiating - I moved IRQ to 0x20, not because it doesn't fit, but because gcc doesn't want it to.
```
picorv32 #(
        .STACKADDR(32'h201000),
        .PROGADDR_RESET(32'h00000000),
        .PROGADDR_IRQ(32'h00000020),
        .ENABLE_IRQ(1),
        .ENABLE_IRQ_QREGS(1),
```

In the link script
```
MEMORY
{
  text    (x!rw) : ORIGIN = 0x00000000, LENGTH = 0x10000
  data    (rw!x) : ORIGIN = 0x00200000, LENGTH = 0x1000
  IO      (rw!x) : ORIGIN = 0x80000000, LENGTH = 0x1000
}

start = 0;

SECTIONS
{
    __stack = 0x201000;

    .text :
    {
        . = 0x00000000;
        *(.startup)
        KEEP (*(.startup))
        . = 0x00000020;
        *(.startirq)
        KEEP (*(.startirq))

        *(.text)
        *(.text.*)
        . = ALIGN(4);
    } > text

    .data :
    {
        __data_start = . ;
        *(.data)
        *(.rela*)
        *(.rodata*)
        __global_pointer$ = . + 0x800;
        *(.srodata .srodata.*)
        *(.sdata .sdata.* .gnu.linkonce.s.*)
        __data_end = . ;
        . = ALIGN(4);
    } > data AT>text

    .bss :
    {
        __bss_start = . ;
        *(.bss)
        *(.dynsbss)
        *(.sbss .sbss.* .gnu.linkonce.sb.*)
        *(.scommon)
        __bss_end = . ;
        start_of_stack = . ;
        _edata = .;
        _end = .;
    }

    __data_load_start = LOADADDR(.data);
}
```
[... 136 lines cut ...]
            "lw x16, 16*4(x1)\n"
            "lw x17, 17*4(x1)\n"
            "lw x28, 28*4(x1)\n"
            "lw x29, 29*4(x1)\n"
            "lw x30, 30*4(x1)\n"
            "lw x31, 31*4(x1)\n");

    picorv32_getq_insn(x1, q2); // restore x1
    picorv32_retirq_insn();
}

void irq(uint32_t irqs)
{
    if (irqs & 1<<0) {
        irq_0();
    }
    if (irqs & (1<<4)) {
        irq_4();
    }
    if (irqs & (1<<5)) {
        irq_5();
    }
}

```

jallennk, 2021-01-05:
I asked this mainly as a check on what GCC does- it apparently does the right thing, it does not use the stack improperly (and there was no need for the Startup.S code to switch to a dedicated interrupt handler stack).  We made a camera using UDP over gigabit Ethernet in a Lattice ECP5 FPGA and used the picorv32 soft-core CPU.  The software is a simple single-task work-queue scheduler with one stack shared by main code and interrupt level code.  All works fine.

closed 2021-01-05 by jallennk