YosysHQ/picorv32 #191: How is the length of the storage area calculated?

itviewer, 2021-03-18:
@stv0g 

https://github.com/cliffordwolf/picorv32/commit/eb64df6c3e5184ee982bb57c8828b80c798ce1b6

in file sections.lds and firmware.c
```
#ifdef ICEBREAKER
#  define MEM_TOTAL 0x20000 /* 128 KB */
#elif HX8KDEMO
#  define MEM_TOTAL 0x200 /* 2 KB */
```
How is this calculated?

void-spark, 2021-04-02:
For ICEBREAKER it's the mem size of the ice40up5k (128kb), which is fully utilized in ice40up5k_spram.v
For HX8KDEMO which doesn't have any memory blocks, it's should be the size of the rtl memory, which you can find at the end of picosoc.v. WORDS is set to 256, which would be 1kb.
Why the hex value is 0x200 (512), and the comment says 2KB, I have no idea, both seem wrong.
The old value in sections.lds seems correct:  0x000400 /* 1 KB */
But there might be a reason for it being this way?

itviewer, 2021-04-03:
I think this is just a mistake