YosysHQ/picorv32 #210: Trap in simulation with existing compiler

pcotret, 2022-01-10:
I was trying to run `make test` with my existing 32-bit compiler, I got the following error:
```
[...]
python3 firmware/makehex.py firmware/firmware.bin 32768 > firmware/firmware.hex
vvp -N testbench.vvp
TRAP after 8224 clock cycles
```
My compiler was configured with:
```
$ ./configure --prefix=/opt/riscv_compiler --with-arch=rv32gc --with-abi=ilp32 --disable-threads
```
Any idea which flag is missing (or maybe several flags) ?

patrickerich, 2022-03-17:
I used the Makefile to install the (rv32i) compiler with:
`RISCV_GNU_TOOLCHAIN_GIT_REVISION = HEAD`
instead of:
`RISCV_GNU_TOOLCHAIN_GIT_REVISION = 411d134`
(This was because I ran into some problems compiling the 411d134 git revision of the toolchain)

I ran into the same TRAP (same number of clock cycles) and it seems to be related to whether or not compressed instructions are used. I just started with this core (so I am not yet very familiar with it), but by changing (in the Makefile):
`COMPRESSED_ISA = C`
to:
`COMPRESSED_ISA = `
I was able to run the `make test`

retrhelo, 2022-05-12:
Just encounter the same problem.

The problem seems to be caused by an unalign memory access in the early part of `firmware/start.S`.

```riscv-assembly
.balign 16
irq_vec:
	/* save registers */

#ifdef ENABLE_QREGS

	picorv32_setq_insn(q2, x1)
	picorv32_setq_insn(q3, x2)

	# Load the address of irq_regs into x1, aka sp
	lui x1, %hi(irq_regs)
	addi x1, x1, %lo(irq_regs)

	picorv32_getq_insn(x2, q0)
	sw x2,   0*4(x1) # This very instruction causes an unaligned memory access
```

However, the symbol `irq_regs` follows right behind some other instructions, and aligned only if `ENABLE_QREGS` is not defined (which is defined by default in the first few lines of `firmware/start.S`). When compiled without compressed instructions, all RISC-V instructions remain 4-byte long, thus `irq_regs` has an alignment of 4-byte. However, when compiling with compressed ones, it's possible for `irq_regs` to align to 2-byte, leading to the unaligned memory access presented above.

```assembly
	lw x28, 28*4+0x200(zero)
	lw x29, 29*4+0x200(zero)
	lw x30, 30*4+0x200(zero)
	lw x31, 31*4+0x200(zero)
#endif

#endif // ENABLE_QREGS

	picorv32_retirq_insn()

#ifndef ENABLE_QREGS
.balign 0x200
#endif
irq_regs:
	// registers are saved to this memory region during interrupt handling
	// the program counter is saved as register 0
	.fill 32,4

	// stack for the interrupt handler
	.fill 128,4
irq_stack:
```

I have no idea why `irq_regs` isn't always set to be aligned. But to solve the problem in this issue, we can simply remove the if-block around `.balign` to make sure that `irq_regs` is always aligned.

AshvinVaidyanathan, 2022-11-16:
I tried removing the ifndef arounf `.balign 0x200` but still doesn't solve the issue. Also tried the solutions Patrick @patrickerich mentioned. The `make test` still throws a 
```
mismatch between q0 LSB and decoded instruction word! q0=0x0000BE78, instr=0x8286
TRAP after 86618 clock cycles
ERROR!
```

patrickerich, 2022-11-16:
I haven't looked at this core for a while, but for me it seems that the problem was related to compiling the RISCV toolchain  from source (although I am not sure about the exact cause).

When using a pre-compiled toolchain the `make test` seems to run without problems. I just tried it (succesfully) with the precompiled toolchain from lowrisc, which you can get [here](https://github.com/lowRISC/lowrisc-toolchains/releases)

If you want to try it, simply:
1) download the [lowrisc-toolchain-rv32imcb-20220524-1.tar.xz](https://github.com/lowRISC/lowrisc-toolchains/releases/download/20220524-1/lowrisc-toolchain-rv32imcb-20220524-1.tar.xz)
2) unpack it somewhere (e.g. /opt/lowrisc)
3) modify the `TOOLCHAIN_PREFIX` variable in the Makefile (e.g. to `TOOLCHAIN_PREFIX = /opt/lowrisc/lowrisc-toolchain-rv32imcb-20220524-1/bin/riscv32-unknown-elf-`

HTH

AshvinVaidyanathan, 2022-11-16:
Hi @patrickerich !

Thanks for the quick reply. i did try the pre-compiled toolchain but it still doesn't work. I made sure to clean the existing files before running it with the new toolchain as well. Still getting the same error.

patrickerich, 2022-11-16:
Hi @AshvinVaidyanathan,

you mentioned:
`I made sure to clean the existing files before running it with the new toolchain as well`
How did you clean the existing files (run a `make clean` or checking out a fresh copy of the git repo)?

As you also mentioned that you tried several things, it would probably be best to start out with a fresh copy of the repo (specifically the Make file) and only modify the `TOOLCHAIN_PREFIX` variable (in the Makefile) to rule out any other changes you may have made (to the Makefile).

patrickerich, 2022-11-17:
Hi @AshvinVaidyanathan,

The solution posted by @retrhelo is completely correct (Thanks @retrhelo!). 
I tried to do a fresh compile of the riscv-gnu-toolchain (eventually even a multilib build) and the solution posted by @retrhelo does indeed solve the issue first mentioned by @pcotret.

I suspect that the reason that the lowrisc pre-build toolchain did not show the same problem (for me), is because of the configuration of the build. I am not sure exactly which option made the difference, but I did notice that a `--disable-libmpx` flag was used in their build.

AshvinVaidyanathan, 2022-11-28:
Hi @patrickerich,

Thanks for your patience. A clean reinstall did the trick.

closed 2023-04-24 by pcotret