YosysHQ/picorv32 #213: Building a RV32I Toolchain: undefined reference to `_initialize_string_view_selftests()'

tommythorn, 2022-03-06:
This strictly isn't a problem with picorv32, but with the RISCV toolchain.  However I'll start here as I suspect many might hit this from here.

Environment: a fully updated Ubuntu 21.10/AMD64

Commands:
```
$ make -j$(nproc) build-tools
This will remove all existing data from /opt/riscv32i, /opt/riscv32ic, /opt/riscv32im, and /opt/riscv32imc.
Type YES to continue: YES
sudo bash -c "set -ex; rm -rf /opt/riscv32{i,ic,im,imc}; mkdir -p /opt/riscv32{i,ic,im,imc}; chown ${USER}: /opt/riscv32{i,ic,im,imc}"
+ rm -rf /opt/riscv32i /opt/riscv32ic /opt/riscv32im /opt/riscv32imc
...
/usr/bin/install -c -m 644 b-header-vars /opt/riscv32i/lib/gcc/riscv32-unknown-elf/8.2.0/plugin/include/b-header-vars
make[4]: Leaving directory '/tmp/picorv32/riscv-gnu-toolchain-riscv32i/build/build-gcc-newlib-stage1/gcc'
make[3]: Leaving directory '/tmp/picorv32/riscv-gnu-toolchain-riscv32i/build/build-gcc-newlib-stage1'
mkdir -p stamps/ && touch stamps/build-gcc-newlib-stage1
make[2]: Leaving directory '/tmp/picorv32/riscv-gnu-toolchain-riscv32i/build'
make[1]: *** [Makefile:158: build-riscv32i-tools-bh] Error 2
make[1]: Leaving directory '/tmp/picorv32'
make: *** [Makefile:167: build-tools] Error 2
```
Digging into this we find
```
/usr/bin/ld: init.o: in function `initialize_all_files()':
/tmp/picorv32/riscv-gnu-toolchain-riscv32i/build/build-gdb-newlib/gdb/init.c:227: undefined reference to `_initialize_string_view_selftests()'
```
The file `/tmp/picorv32/riscv-gnu-toolchain-riscv32i/build/build-gdb-newlib/gdb/init.c` is auto-generated by the GDB Makefile and the relevant file exist but is empty:
```
$ objdump -d ./riscv-gnu-toolchain-riscv32i/build/build-gdb-newlib/gdb/unittests/string_view-selftests.o

./riscv-gnu-toolchain-riscv32i/build/build-gdb-newlib/gdb/unittests/string_view-selftests.o:     file format elf64-x86-64

```

I've tried various hacks, but as everything is overwritten by git, and much is autogenerated, this hasn't been successful yet.  I'm throwing up this issue in case somebody has hit this and have a suggestion.

usbalex, 2022-03-17:
I'm having the same issue on Debian bookworm/sid.
```
$ make
...
  GEN    init.c
  CXX    init.o
  CXXLD  gdb
/usr/bin/ld: init.o: in function `initialize_all_files()':
<my_path>/picorv32/riscv-gnu-toolchain-rv32i/build/build-gdb-newlib/gdb/init.c:227: undefined reference to `_initialize_string_view_selftests()'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:1910: gdb] Error 1
make[3]: Leaving directory '<my_path>/picorv32/riscv-gnu-toolchain-rv32i/build/build-gdb-newlib/gdb'
make[2]: *** [Makefile:9126: all-gdb] Error 2
make[2]: Leaving directory '<my_path>/picorv32/riscv-gnu-toolchain-rv32i/build/build-gdb-newlib'
make[1]: *** [Makefile:849: all] Error 2
make[1]: Leaving directory '<my_path>/picorv32/riscv-gnu-toolchain-rv32i/build/build-gdb-newlib'
make: *** [Makefile:416: stamps/build-gdb-newlib] Error 2
```
My gdb version is:
```
$ gdb --version
GNU gdb (Debian 10.1-2) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

tommythorn, 2022-03-17:
Sorry, I failed to update with my working workaround.  It's not pretty, but it works.  First, we have to inject at a point in the build process:
```
diff --git a/Makefile b/Makefile
index d7027e3..083ea40 100644
--- a/Makefile
+++ b/Makefile
@@ -150,7 +150,7 @@ build-$(1)-tools-bh:
        git submodule update --init $$$$reference_riscv_gcc riscv-gcc; \
        git submodule update --init $$$$reference_riscv_glibc riscv-glibc; \
        git submodule update --init $$$$reference_riscv_newlib riscv-newlib; \
-       mkdir build; cd build; ../configure --with-arch=$(2) --prefix=$(RISCV_GNU_TOOLCHAIN_INSTALL_PREFIX)$(subst riscv32,,$(1)); make
+       mkdir build; cd build; ../configure --with-arch=$(2) --prefix=$(RISCV_GNU_TOOLCHAIN_INSTALL_PREFIX)$(subst riscv32,,$(1)) #; make
```

tommythorn, 2022-03-17:
at which point I could remove the offending line from picorv32/riscv-gnu-toolchain-riscv32i/riscv-gdb/Makefile.in:
```
diff --git a/gdb/Makefile.in b/gdb/Makefile.in
index 3be058f052..bff8d275a3 100644
--- a/gdb/Makefile.in
+++ b/gdb/Makefile.in
@@ -425,7 +425,6 @@ SUBDIR_UNITTESTS_SRCS = \
        unittests/scoped_fd-selftests.c \
        unittests/scoped_mmap-selftests.c \
        unittests/scoped_restore-selftests.c \
-       unittests/string_view-selftests.c \
        unittests/tracepoint-selftests.c \
        unittests/unpack-selftests.c \
        unittests/utils-selftests.c \
```

I didn't follow up because I wanted a cleaner fix, but hopefully this can help someone.

usbalex, 2022-03-17:
Thanks @tommythorn ! I could get it working this way

tyler274, 2022-09-29:
Also getting this problem.

tommythorn, 2024-11-04:
I forgot how much I hated building the GNU tools.  The description I gave appears inexact.  This is what I did that worked for me:
```
$ git clone https://github.com/YosysHQ/picorv32.git
$ cd picorv32
$ make download-tools
$ make build-tools # followed by typing YES
... the build eventually fails, complaining about undefined reference to `_initialize_string_view_selftests()'
$ find . -name Makefile.in|xargs grep unittests/string_view-selftests.c -l|xargs sed -i -e '/unittests\/string_view-selftests.c/d'
$ make -C riscv-gnu-toolchain-riscv32i/build
```

I just tested it from scratch on an current Ubuntu 24.04.1 LTS and it Worked For Me AFAICT™

tommythorn, 2024-11-09:
Sorry, I don't have time to work on this further.  If I did, I would investigate why we even need to build the tools given that most modern distributions ship with cross tools that are perfectly fine.  I assume it's because we need newlib, but it should be possible to build this with stock tools.

Regardless, it seems apparent that this repo has been abandoned.

alex-strangelove, 2025-04-26:
The same issue on Ubuntu 22.04/24.04.

If you don't need GDB, here is workaround (skip building GDB):

`picorv32/Makefile`:
```
build-$(1)-tools-bh:
        ...
	mkdir build; cd build; ../configure --with-arch=$(2) --disable-gdb --prefix=$(RISCV_GNU_TOOLCHAIN_INSTALL_PREFIX)$(subst riscv32,,$(1)); make
```