YosysHQ/picorv32 #214: Toolchain:

Riesi, 2022-03-09:


Riesi, 2022-03-09:
ups hit ctrl enter by accident

Riesi, 2022-03-09:
Am I missing something? Why this repo is building different compilers per each architecture instead of simply using -march

jeras, 2022-08-21:
I would too like to be able to test PicoRV32 (`make test_verilator`) with a prebuilt compiler, my choice would be a nightly from https://github.com/riscv-collab/riscv-gnu-toolchain

I first modified the Makefile to accept a system variable specyfiing the compiler, but I could not get past library issues (either floating point or more general for `riscv64-elf-ubuntu-20.04-nightly-2022.08.19-nightly`).
```
$ git diff Makefile 
diff --git a/Makefile b/Makefile
index d7027e3..dcd1b04 100644
--- a/Makefile
+++ b/Makefile
@@ -15,7 +15,7 @@ TEST_OBJS = $(addsuffix .o,$(basename $(wildcard tests/*.S)))
 FIRMWARE_OBJS = firmware/start.o firmware/irq.o firmware/print.o firmware/hello.o firmware/sieve.o firmware/multest.o firmware/stats.o
 GCC_WARNS  = -Werror -Wall -Wextra -Wshadow -Wundef -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings
 GCC_WARNS += -Wredundant-decls -Wstrict-prototypes -Wmissing-prototypes -pedantic # -Wconversion
-TOOLCHAIN_PREFIX = $(RISCV_GNU_TOOLCHAIN_INSTALL_PREFIX)i/bin/riscv32-unknown-elf-
+TOOLCHAIN_PREFIX ?= $(RISCV_GNU_TOOLCHAIN_INSTALL_PREFIX)i/bin/riscv32-unknown-elf-
 COMPRESSED_ISA = C
 
 # Add things like "export http_proxy=... https_proxy=..." here
```