YosysHQ/picorv32 #223: Bug: TRAP after >8000 clock cycles

kaminto-1999, 2023-01-11:
I have tried to run all the "make test*" in the root folder.
But seem like the "make test*" command returns an Error after the run for about 8000 cycles.
Does anyone know how to fix it?
Thank you.
[image]

filamoon, 2023-02-17:
Same here. No error messages at all.

TurBoss, 2023-08-29:
hello, i managed to get it build by using this command

` make TOOLCHAIN_PREFIX=riscv32-unknown-elf-`

:)

rigoorozco, 2023-11-03:
I'm getting the same thing. I was able to get around it using version 10.2.0 here:
https://github.com/xpack-dev-tools/riscv-none-embed-gcc-xpack/releases/

Not sure what the problem is but it seems to not work with newer versions of gcc

rigoorozco, 2023-11-03:
Got around this issue on GCC 13.2.0 doing the following to the Makefile:
```
diff --git a/Makefile b/Makefile
index d7027e3..feda8a7 100644
--- a/Makefile
+++ b/Makefile
@@ -15,8 +15,8 @@ TEST_OBJS = $(addsuffix .o,$(basename $(wildcard tests/*.S)))
 FIRMWARE_OBJS = firmware/start.o firmware/irq.o firmware/print.o firmware/hello.o firmware/sieve.o firmware/multest.o firmware/stats.o
 GCC_WARNS  = -Werror -Wall -Wextra -Wshadow -Wundef -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings
 GCC_WARNS += -Wredundant-decls -Wstrict-prototypes -Wmissing-prototypes -pedantic # -Wconversion
-TOOLCHAIN_PREFIX = $(RISCV_GNU_TOOLCHAIN_INSTALL_PREFIX)i/bin/riscv32-unknown-elf-
-COMPRESSED_ISA = C
+TOOLCHAIN_PREFIX ?= $(RISCV_GNU_TOOLCHAIN_INSTALL_PREFIX)i/bin/riscv32-unknown-elf-
+COMPRESSED_ISA = 
 
 # Add things like "export http_proxy=... https_proxy=..." here
 GIT_ENV = true
@@ -116,7 +116,7 @@ firmware/start.o: firmware/start.S
        $(TOOLCHAIN_PREFIX)gcc -c -mabi=ilp32 -march=rv32im$(subst C,c,$(COMPRESSED_ISA)) -o $@ $<
 
 firmware/%.o: firmware/%.c
-       $(TOOLCHAIN_PREFIX)gcc -c -mabi=ilp32 -march=rv32i$(subst C,c,$(COMPRESSED_ISA)) -Os --std=c99 $(GCC_WARNS) -ffreestanding -nostdlib -o $@ $<
+       $(TOOLCHAIN_PREFIX)gcc -c -mabi=ilp32 -march=rv32im$(subst C,c,$(COMPRESSED_ISA)) -Os --std=c99 $(GCC_WARNS) -ffreestanding -nostdlib -o $@ $<
 
 tests/%.o: tests/%.S tests/riscv_test.h tests/test_macros.h
        $(TOOLCHAIN_PREFIX)gcc -c -mabi=ilp32 -march=rv32im -o $@ -DTEST_FUNC_NAME=$(notdir $(basename $<)) \
```

KrystalDelusion, 2024-03-26:
Closed by #218

closed 2024-03-26 by KrystalDelusion