YosysHQ/picorv32 #233: What is PCPI base address?

suryacharanp, 2023-03-08:
Hi,  
Does anyone know what is the base address of PCPI interface? I  am changing firmware to access my PCPI based co-processor from software but could not find its base address. I dont know if I am overseeing something basic here. Thanks in advance.

gatecat, 2023-03-08:
There is no base address for PCPI, it's a way of adding custom instructions not memory mapped peripherals.

suryacharanp, 2023-03-09:
Hi gatecat, thanks for clearing my doubt, do you know any example where pcpi is controlled from software? that would be very helpful, I checked for the examples like fast_mul but I couldn't find any traces of pcpi  in firmware?

gatecat, 2023-03-09:
In this case, `fast_mul` is just implementating and handling the standard RISC-V multiply instructions.

suryacharanp, 2023-03-11:
Thanks for the clarification @gatecat , do you have any idea about the structure of opcode one must use to access pcpi?

suryacharanp, 2023-06-09:
You can get the base from the examples given for the pcpi, the bits where you cant find it..just make them 0...it will work

closed 2023-06-09 by suryacharanp