YosysHQ/picorv32 #246: wbm_sel_o == 0 during read transactions

epsilon537, 2023-09-19:
The picorv32_wb state machine always sets wbm_sel_o to 0 during Wishbone read transactions:

					if (mem_valid) begin
						wbm_adr_o <= mem_addr;
						wbm_dat_o <= mem_wdata;
						wbm_we_o <= we;
						wbm_sel_o <= mem_wstrb;

During read transactions mem_wstrb will be 0, so wbm_sel_o gets set to 0. According to the Wishbone spec, sel_o also applies to read transactions, however. Shouldn't wbm_sel_o be set to 4'b1111 during read transactions?

marketideas, 2024-12-25:
I just ran into this as well. I can't believe no one else has found this. Trying to implement the OpenCores 16550 uart, and it requires wb_sel_o to be set during reads so it can properly return an 8 bit register (8 bit aligned) into a 32 bit data width.

GrobiG, 2025-03-24:
Yesterday I encountered the same issue, (picorv32 IP from gowin IDE) byte-wise writing via the openwishbone interface works as expected but when reading the sel signal is always "0000". Thought I was doing something wrong and spent the whole day wrestling with my design and c code :) Is there a fix for this? Does this work via the openahb bus?